arduino/Arduino #7633: [Library Manager] Please add ACAN library

pierremolinaro, 2018-05-25:
Hi,

Please, can you include my ACAN library (CAN driver for Teensy 3.x):
https://github.com/pierremolinaro/acan/tree/master/teensyduino-library/ACAN

Thanks,

Pierre Molinaro

per1234, 2018-05-25:
In order to be added to the Library Manager index, your library must be located in the root of the repository, rather than in the teensyduino-library/ACAN subfolder.

Please resolve this issue, make a new release or tag, then comment here so we know we can proceed with adding your repository to the Library Manager index.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

pierremolinaro, 2018-05-25:
Ok, sorry about that. I do the modification.

Thanks,

Pierre Molinaro

pierremolinaro, 2018-05-25:
It is done. The ACAN library is the repository: https://github.com/pierremolinaro/acan

Thanks,

Pierre Molinaro

cmaglie, 2018-05-25:
Done

closed 2018-05-25 by cmaglie

pierremolinaro, 2018-05-26:
Thank you very much !

Pierre Molinaro