arduino/Arduino #7636: Parent sketch directory not recognized when given relative path. Workaround.

mirage335, 2018-05-26:
Path "/home/user/sketch/sketch.ino" is accepted. Path "./sketch.ino" results in an error, as the sketch is not recognized to reside in parent directory "sketch".

Translating all paths to absolute does work, as demonstrated by the linked script.

https://github.com/mirage335/arduinoUbiquitous/blob/master/_prog/core.sh#L109

facchinm, 2018-05-28:
Hi @mirage335 ,
which IDE version are you using? I just tested with 1.9.0 (Beta) and it correctly compiles  sketches specified with relative paths.

mirage335, 2018-05-29:
Interesting, will have to try the beta. Was using version 1.8.5, with SAMD core 1.6.18 and 1.6.16-build-172, as currently contained by arduinoUbiquitous .

So, maybe not a bug with the current Arduino codebase. Then again, it is also possible this is a java issue. In either case, file parameter expansion could still be a useful workaround with older versions or java runtimes.

Thanks for looking into this.

mirage335, 2018-05-30:
Same error with beta and nightly. Additionally, relative paths simply do not work with Debian's built-in Arduino installation, opening a blank sketch, without error message.

Perhaps it may be significant that I am using Debian Stable.

cmaglie, 2018-05-30:
I can't reproduce it locally:

```
~$ wget https://downloads.arduino.cc/arduino-nightly-linux64.tar.xz
~$ tar xf arduino-nightly-linux64.tar.xz
~$ mkdir sketch
~$ echo "void setup() {} void loop() {}" > sketch/sketch.ino
~$ arduino-nightly/arduino --verify sketch/sketch.ino 
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Sketch uses 48432 bytes (31%) of program storage space. Maximum is 155648 bytes.
~$ cd sketch
~/sketch$ ../arduino-nightly/arduino --verify ./sketch.ino 
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Sketch uses 48432 bytes (31%) of program storage space. Maximum is 155648 bytes.
~/sketch$ 

```

 could you provide a step-by-step sequence of command to reproduce it?

mirage335, 2018-05-31:
Try running just "../arduino-nightly/arduino ./sketch.ino" . If that works, this may be a Debian specific issue.