arduino/Arduino #7638: IDE - text selection improvements

aster94, 2018-05-26:
hi guys,

since you decided that you want a real IDE with debugging (I immagine it is planned in 1.9 right?) I guess it is time to add some functionalities as the other "cool" IDE out there have

- multiline cursor
- multiple selection
- multiple panels

I did a short video but I am pretty sure that everyone knows what I am talking about
https://streamable.com/i7gmz
I guess this would be a good starting point

aster94, 2018-06-10:
@per1234 do you think the label "ide 1.9" should be added?

per1234, 2020-10-05:
>multiline cursor

If I understand what this means, we already have a feature request: https://github.com/arduino/Arduino/issues/6425

per1234, 2021-03-02:
Great news everyone! This requested functionality is now available in [Arduino IDE 2.x](https://github.com/arduino/arduino-ide). By default, it uses the same standard keyboard shortcut as VS Code and Notepad++.

It's still in the beta development phase, but if you like, you're welcome to try it out:
https://www.arduino.cc/en/software#experimental-software

closed 2021-03-02 by per1234