arduino/Arduino #7641: Installation issue in arduino IDE in ubuntu

NishanthARao, 2018-05-28:
Hi. While the instructions on the main website on installing arduino on linux systems (Specifically ubuntu) provide a comprehensive guide, this may not work for all (E.g., myself!).
You may get some errors like (........../ No such file or directory)
The problem lies in the directory from where you run the ./install.sh file. For some users, the installation can work without any problems when extracted into downloads folder itself, but for me i had to put the arduino-(version) folder (that contains the script ./install.sh) seperately onto the home directory (that includes folders like Desktop, Documents, Downloads, etc) and then 'cd' into arduino folder and then run the script. This solves the error.

matthijskooijman, 2018-05-28:
Can you please be more explicit what errors you are seeing? We can't figure out what is wrong, let alone fix it, based on just the workaround you found.

NishanthARao, 2018-05-28:
Hi.

I m unable to recollect exactly what the errors where. But it was something
like this:

Cannot create home/(username)/.local/...../(filename).(extension) No such
file of directory.

I think, in the script, you would have written where the files have to be
copied relative to the current directory from where the ./install.sh was
run. This caused an error(Maybe my system problem) as it couldn't 'cd' into
proper directories and copy the main files

matthijskooijman, 2018-05-28:
Your partial error suggests there was a problem in one of the `cp` commands here: https://github.com/arduino/Arduino/blob/0a84ac21387abc92f63a56ec920f4b7e4b3d0520/build/linux/dist/install.sh#L88-L98 (in particular, the directory copied to would not exists). However, each of these either createst the target directory, or only runs if it exists.

Based on the info you gave, there is nothing we can really do to fix this, so I'm closing this. If you want to provide more info (e.g. by trying to run the script again), feel free to add more info and reopen this issue.

closed 2018-05-28 by matthijskooijman

NishanthARao, 2018-05-28:
Thanks a lot for your support.