arduino/Arduino #7651: Use proper configuration files location on linux

Perlovka, 2018-05-30:
Following https://github.com/arduino/Arduino/issues/7635

It would be nice to change config directory to $HOME/.config/arduino, to comply with [XDG Base Directory Specification](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html#variables) ( $XDG_CONFIG_HOME )

per1234, 2018-05-31:
Closing as duplicate of https://github.com/arduino/Arduino/issues/3915

We welcome bug reports and feature requests but in the future please take the time to search for existing open and closed issues and pull requests before submitting an issue:
https://github.com/arduino/Arduino/issues?&q=
If you do find the issue has already been reported or a fix has been submitted, you are welcome to give it a "thumbs up" by clicking on the smiley face icon at the top right to show your support. Please only comment on the issue if you have additional information to add.

closed 2018-05-31 by per1234

Perlovka, 2018-05-31:
The reason previous bug report been closed with, shows how really bugs are welcome. And there is no reopen functionality, so new bug been filled for this purpose.

per1234, 2018-05-31:
Legitimate, unique issue reports are certainly welcome here. They are a valuable contribution to the Arduino project.

GitHub only allows repository administrators or the person who closed an issue to re-open, and for good reason. If you want a closed issue to be re-opened then you need to comment on that issue with new information that makes it clear why it should be re-opened. Note that unproductive comments on an issue may lead to its comment thread being locked, which will certainly not be helpful to your goal.

It is absolutely not appropriate to open duplicate issue reports. This repository already has >6400 issue reports. Duplicate and invalid issues make maintaining and searching so many issues even more difficult.

PaulStoffregen, 2018-05-31:
This really depends on how you define "nice".

Complying with the freedesktop spec might be "nice".  But what is the real benefit?

Making this change may not be so "nice" for countless users who already have quite a lot of custom config accumulated in ~/.arduino15.  Updating numerous web pages that mention the existing location doesn't seem like such a "nice" task, and especially for those located on non-Arduino websites.  Extra work to automatically migrate people's data would be "nice", but what happens if those users have more than 1 installation of Arduino on their machine?  (yes, that's a very common scenario, where people begin a hardware project using an older version and keep a copy around for only that project)

There are a lot of ways for this change to be not so "nice" in practice, which is probably why the proposal was rejected.  I personally think it's a bad idea, with lots of compatibility & documentation headaches for end users.  So what is the rationale, what is the compelling reason to need this?  It better have a lot more real-world benefit that merely "nice"!

feikname, 2018-06-01:
> Complying with the freedesktop spec might be "nice". But what is the real benefit? How does it actually help people?

It goes in compliance with the commonly used standard rather than being yet another `.dotApp` in users's home folder. It also allows for easily having multiple versions of Arduino each with their own settings by having a custom `$XDG_CONFIG_HOME` when launching it.

Having said that, I do believe the change is not worth it too.

Perlovka, 2018-06-01:
Well, first of all, I'm sorry for my rude comment, it was not "nice" at all ))
I just don't like when bugs are closed with the reason "We just won't do this". If it were closed with an explanation like this one from Paul, that would be just perfect =)

As for moving config location to another location, it's actually not such a big problem, IMO. Many projects moved their default config location to comply with the standards. For example, upon updating to next major release (1.9) just warn the user that from this release default config directory will be located in new place.

No need in automating that process. And current location should not be disabled immediately.

So the logic of my point of view should be like this:
```
if exists $XDG_CONFIG_HOME/.arduino {
    use it
}
else if exists ~/.config/arduino {
    use it
}
else if exists ~/.arduino15 {
    use it
}
else {
    create $XDG_CONFIG_HOME/.arduino or ~/.config/arduino
}
```

In such a way no hard work should be done, and there will be standard way to keep multiple installations.
Thanks.

RockyTV, 2018-08-14:
> Updating numerous web pages that mention the existing location doesn't seem like such a "nice" task, and especially for those located on non-Arduino websites.

If the docs on these pages are updated for every new version, why would a single directory rename bother at all?

> but what happens if those users have more than 1 installation of Arduino on their machine? (yes, that's a very common scenario, where people begin a hardware project using an older version and keep a copy around for only that project)

They would have different folders, different configs, different settings. It's an advantage, not a problem. E.g. you'd have `~/.config/arduino15` and `~/.config/arduino16`.

> There are a lot of ways for this change to be not so "nice" in practice, which is probably why the proposal was rejected.

The way I see it is that the developers don't even want to think of a way of dealing with the "headache", not allowing the users to choose where they want to store their files. Yes, I agree that using XDG directories by default would upset a lot of users. But why not give us the option to change the directories, instead of hardcoding them?
  
  
If you need more reasons, I'll quote from https://rubyworks.github.io/xdg/ page:

>### What is this all about?
>
>The XDG Base Directory Standard provides a common means for specifying one or more base directories relative to which support files can be located. Using the XDG standard, applications can find and store application specific support files (eg. config files, data caches, etc.) in a universal, clean and robust fashion. The advantages of this standard is that all such files will be found in central locations easily discernible by man and machine. And, if widely adopted, XDG can clear up our HOME directories of all those pesky hidden files.
>
> ### Why should I use the XDG standard?
>
>If your program depends on user-relative or system-wide support files, it will be much improved by using the XDG Base Directory Standard. Not only does it provide a clean yet flexible standard, it also makes your life easier. The `#find` and `#select` methods make it a snap to locate support files.
>
>XDG make it easy to enable your application's users to create user-centric or system-centric configuration and support files, and store those files where they see fit --whether in the standard locations or locations of personal preference.
>
> ### What can XDG standard ultimately do?
>
>How about turning this:
>
>    ```
>    $ ls -a ~/
>
>    .ICEauthority     .gconf                .qt                    
>    .Xauthority       .gconfd               .rdoc                  Documents
>    .Xsession         .gem                  .recently-used         Download
>    .adobe            .gftp                 .recently-used.xbel    EMail
>    .aptitude         .gimp-2.4             .ri                    Friends
>    .aspell.en.prepl  .gitconfig            .ruby_inline           Jobs
>    .aspell.en.pws    .gksu.lock            .rubyrc                Mail
>    .bash_history     .gnome                .sane                  Manifest.txt
>    .bash_logout      .gnome2               .shoes                 Media
>    .bashrc           .gnome2_private       .ssh                   Memento
>    .bogofilter       .gnupg                .subversion            Music
>    .cache            .gstreamer-0.10       .themes                Notebook
>    .camel_certs      .gtk-bookmarks        .thumbnails            Personal
>    .compiz           .gtk-recordmydesktop  .tremulous             Photos
>    .config           .gtkrc-1.2-gnome2     .update-manager-core   Pics
>    .dbus             .gvfs                 .update-notifier       Pictures
>    .dbus-keyrings    .icons                .vim                   PonyArchive
>    .dia              .irb_history          .viminfo               Public
>    .diakonos         .irbrc                .vimrc                 Sort
>    .dmrc             .local                .vlc                   Templates
>    .dvdcss           .macromedia           .w3m                   Videos
>    .dvdrip           .metacity             .wapi                  current
>    .dvdrip-master    .mozilla              .xine                  dvdrip-data
>    .dvdriprc         .mplayer              .xinitrc               links.xml
>    .eee              .mysql_history        .xsession-errors       programs
>    .esd_auth         .nautilus             Archive                projects
>    .etc              .openoffice.org2      Art                    ruby
>    .evolution        .pgadmin3             Bookmarks              sandbox
>    .fastri-index     .pgpass               Books                  tags
>    .fehbg            .profile              Company                tigerops
>    .fehrc            .pulse                Contacts
>    .fontconfig       .pulse-cookie         Desktop
>    .fr-KVIe0s        .purple               Diff.txt
>    ```
>
>Into this:
>
>```
>    $ ls -a ~/
>
>    .cache            Diff.txt              Music                  Videos
>    .config           Documents             Notebook               current
>    .local            Download              Personal               dvdrip-data
>    Archive           EMail                 Photos                 links.xml
>    Art               Friends               Pics                   programs
>    Bookmarks         Jobs                  Pictures               projects
>    Books             Mail                  PonyArchive            ruby
>    Company           Manifest.txt          Public                 sandbox
>    Contacts          Media                 Sort                   tags
>    Desktop           Memento               Templates              tigerops
>````
>

rouing, 2018-12-27:
This really needs to be revisited. Decluttering the userspace is a top priority and being stubborn about the change is just being stubborn for no good. This needs to catch up with the standards or be forked imo.

gahag, 2018-12-28:
Writing efficient backup scripts for the home directory is a nightmare due to this lack of basic organisation. I must exclude every single application specific folder from my backup routine, instead of just excluding the ~/.config and ~/.local directories. I don't want to backup the arduino configs. I want to back up my personal files. Following the standard is a good practice, and a sign of good engineering. This issue should be definitely reconsidered.

sparr, 2019-02-03:
> > Updating numerous web pages that mention the existing location doesn't seem like such a "nice" task, and especially for those located on non-Arduino websites.

> If the docs on these pages are updated for every new version, why would a single directory rename bother at all?

Taking this a step further... Making this change would make it obvious to at least some users when the documentation they are reading is out of date. That's a healthy sort of change to make every decade or so.

ifohancroft, 2019-02-19:
Another benefit of starting to comply with the standard is that it will certainly improve the reputation of the Arduino Foundation in the eyes of a big part of the community. Of course, as long as it is The Foundation that sits behind the Arduino software itself and its repo.

Perlovka, 2019-10-03:
If anybody is still interested, i've created a patch to fix this issue.
And this https://github.com/arduino/Arduino/issues/3915
Also, https://github.com/arduino/Arduino/issues/7635 should be fixed as well.

The new bihavior is as follows:
If $HOME/.arduino15 exists, it will be used as config directory.
Else, if XDG_CONFIG_HOME is set, $XDG_CONFIG_HOME/arduino15 will be used.
Finally - $HOME/.config/arduino15 will be used.

Additionally, cache directory has been moved:
If XDG_CACHE_HOME is set, then $XDG_CACHE_HOME/arduino will be used
Else - $HOME/.cache/arduino

The patch can be found here:
https://github.com/Perlovka/portage-overlay/blob/master/dev-embedded/arduino/files/arduino-1.8.10-xdg-compliance.patch

P.S. There is invisible holy finger of Linus in this thread. A big fat finger.

lhanson, 2019-10-17:
@Perlovka Has this been submitted as a pull request?

Perlovka, 2019-10-17:
@lhanson No, this is "wontfix" issue. I have no intention to make a PR that will be closed like all these issues. But if you are brave anough - you are welcome )

matthijskooijman, 2019-10-17:
This issue has been closed as a duplicate of #3915, so further discussion, if any, would better happen there.