arduino/Arduino #7653: Re: wrong jpeg header with OV5642 camera shield (Arducam-M-5MP)

arduino123456, 2018-05-31:
I am using an Arducam Mini module camera shield 5MP plus OV5642 with ESP8266 UNO board. I am trying to capture an image and write to SD card in JPEG mode.

Attached is the sketch that I am using. I added the "camera_debug" function to debug the issue. Here are the list of items I checked.

a.) SPI interface - ok. Using SPI_CS 16
b.) SD card SD_CS is 0. Checked that I can write and read a simple pattern to the SD card
c.) OV5642 module is getting detected based on the messages on the serial monitor. However, no image is written to the SD card. When I tried to print the JPEG header bytes on the serial monitor, they were 0xBFD8 instead of 0xFFD8. ``

I am using Arduino IDE 1.8.5. Any pointers greatly appreciated. Thanks.

//This demo can only work on ARDUCAM_SHIELD_V2  platform.
#if !(defined (ARDUCAM_SHIELD_V2)&&(defined (OV5640_CAM) ||defined (OV5642_CAM)||defined (OV2640_CAM)))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#if defined(ESP8266)
 #define SD_CS 0 
 const int SPI_CS = 16;
#else 
 #define SD_CS 9 
 const int SPI_CS =10;
#endif
#if defined (OV2640_CAM)
  ArduCAM myCAM(OV2640, SPI_CS);
#elif defined (OV5640_CAM)
  ArduCAM myCAM(OV5640, SPI_CS);
#elif defined (OV5642_CAM)
  ArduCAM myCAM(OV5642, SPI_CS);
#endif
void myCAMSaveToSDFile(){
char str[8];
byte buf[256];
static int i = 0;
static int k = 0;
uint8_t temp = 0, temp_last = 0;
uint32_t length = 0;
bool is_header = false;
File outFile;
//Flush the FIFO
myCAM.flush_fifo();
//Clear the capture done flag
myCAM.clear_fifo_flag();
//Start capture
myCAM.start_capture();
Serial.println(F("Start Capture"));
while(!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));
Serial.println(F("Capture Done."));  
length = myCAM.read_fifo_length();
Serial.print(F("The fifo length is :"));
Serial.println(length, DEC);
if (length >= MAX_FIFO_SIZE) //8M
{
  Serial.println(F("Over size."));
  return ;
}
if (length == 0 ) //0 kb
{
  Serial.println(F("Size is 0."));
  return ;
}
//Construct a file name
k = k + 1;
itoa(k, str, 10);
strcat(str, ".jpg");
//Open the new file
outFile = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
if(! outFile){
  Serial.println(F("File open faild"));
  return;
}
i = 0;
[... 247 lines cut ...]
#endif
myCAM.set_format(JPEG);
myCAM.InitCAM();
delay(100);
#if defined (OV2640_CAM)
myCAM.OV2640_set_JPEG_size(OV2640_320x240);delay(1000);
#elif defined (OV5640_CAM)
myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
myCAM.OV5640_set_JPEG_size(OV5640_320x240);delay(1000);
#elif defined (OV5642_CAM)
//myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
myCAM.set_bit(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);
myCAM.clear_fifo_flag();
myCAM.write_reg(ARDUCHIP_FRAMES,0x00);
myCAM.OV5642_set_JPEG_size(OV5642_640x480);delay(1000);
#endif
delay(1000);
camera_debug();
delay(5000);
}

void loop(){
//  myCAMSaveToSDFile();
//  delay(5000);
}

per1234, 2018-05-31:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum, where I see you've already posted:
http://forum.arduino.cc/index.php?topic=550528
I'm sure we'll be able to help you with your problem over there.

closed 2018-05-31 by per1234

arduino123456, 2018-05-31:
ok. Thanks.