arduino/Arduino #7654: Changing library name: MoliCAN --> ACAN

pierremolinaro, 2018-05-31:
Hello,

I have submitted my library last saturday (https://github.com/pierremolinaro/acan), and I realized I had the wrong name in the "name" field. I have made the correction (MoliCAN --> ACAN), changed the version to 1.0.2 and made a release.

But there is no change in library list, perhaps I have done something wrong ?

Best Regards,

Pierre Molinaro

per1234, 2018-05-31:
The library name is locked in at the time it's added to the index so it needs to be manually changed by one of the Arduino team. cmaglie will make the requested name change on their next run through the Library Manager inclusion requests and close this issue at that time.

pierremolinaro, 2018-05-31:
Thank you very much.

Sorry for the additional work.

Best regards,

Pierre Molinaro

cmaglie, 2018-06-05:
done

closed 2018-06-05 by cmaglie