arduino/Arduino #7655: Error in arduino library

Ooip83, 2018-05-31:
What does these mean?
............................................................................................................................................................................................................
Arduino: 1.8.3 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Users\OOIYAN~1\AppData\Local\Temp\ccyBBxge.ltrans0.ltrans.o: In function `__base_ctor ':

C:\Users\ooi yan liang\Documents\Arduino\libraries\CheapStepper-master/CheapStepper.cpp:31: undefined reference to `pinMode'

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/lib/avr5/crtatmega328p.o:(.init9+0x0): undefined reference to `main'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
...................................................................................................................................................................................................

Ooip83, 2018-05-31:
I really need help

schinken, 2018-05-31:
In my opinion your ticket should be in this repo: https://github.com/tyhenry/CheapStepper because its an error in the external library

per1234, 2018-05-31:
I don't see a problem in the library. I believe this is caused by a problem with the Arduino IDE's compile cache. @Ooip83 please close all Arduino IDE windows and then restart it and try again. That should clear the cache and hopefully make the errors go away. Please let us know what the results are.

Ooip83, 2018-06-01:
Yes. It works.

Ooip83, 2018-06-01:
Thanks anyway

closed 2018-06-01 by Ooip83