arduino/Arduino #7666: the source code split on several tabs didn't compile on 1.9 beta

artua, 2018-06-04:
Multiple 'somefunction' was not declared in this scope errors.
1.8.5 and 1.8.6 nightly build compiles fine.

facchinm, 2018-06-04:
Hi @artua ,
could you share the sketch so we can try reproducing the issue? Thanks

artua, 2018-06-04:
It's 2000 lines of code. I'll try to create MCVE

artua, 2018-06-04:
@facchinm ok, I created MCVE. It doesn't relates on tabs. It relates on where I put my functions (they was on tabs) and also depends on some other things.

If I delete any line from PROGMEM definitions or remove an include, it compiles without error.
If I move void logger(String str) definition before setup() - it compiles again.

It compiles without errors on 1.8.5 as I stated before.

**ERROR LOG**
```
Compiling sketch...
"/Users/sierrarc/Documents/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/sierrarc/Documents/Arduino/hardware/esp8266com/esp8266/tools/sdk/include" "-I/Users/sierrarc/Documents/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/include" "-I/Users/sierrarc/Documents/Arduino/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/var/folders/2w/zv6zwl615y30r6qn_qfgz1c00000gn/T/arduino_build_261056/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DDEBUG_ESP_PORT=Serial  -DARDUINO=10900 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-I/Users/sierrarc/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266" "-I/Users/sierrarc/Documents/Arduino/hardware/esp8266com/esp8266/variants/generic" "-I/Users/sierrarc/Documents/Arduino/libraries/ArduinoJson/src" "/var/folders/2w/zv6zwl615y30r6qn_qfgz1c00000gn/T/arduino_build_261056/sketch/wm-bug1.9.ino.cpp" -o "/var/folders/2w/zv6zwl615y30r6qn_qfgz1c00000gn/T/arduino_build_261056/sketch/wm-bug1.9.ino.cpp.o"
/var/folders/2w/zv6zwl615y30r6qn_qfgz1c00000gn/T/arduino_modified_sketch_942930/wm-bug1.9.ino: In function 'void setup()':
wm-bug1.9:18:28: error: 'logger' was not declared in this scope
   logger("Startup timing: ");
                            ^
Using library ArduinoJson at version 5.13.1 in folder: /Users/sierrarc/Documents/Arduino/libraries/ArduinoJson 
exit status 1
'logger' was not declared in this scope
```
**Code follows**
```
#include <ArduinoJson.h>

const char HTTP_STYLE[] PROGMEM           = "<style>.sm{font-size:0.8em;}.c{text-align: center;} div,input{padding:5px;font-size:1em;} input{width:95%;} body{text-align: center;font-family:verdana;} button{border:0;border-radius:0.3rem;background-color:#1fa3ec;color:#fff;line-height:2.4rem;font-size:1.2rem;width:100%;} .q{float: right;width: 64px;text-align: right;} .l{background: url(\"data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAACAAAAAgCAMAAABEpIrGAAAALVBMVEX///8EBwfBwsLw8PAzNjaCg4NTVVUjJiZDRUUUFxdiZGSho6OSk5Pg4eFydHTCjaf3AAAAZElEQVQ4je2NSw7AIAhEBamKn97/uMXEGBvozkWb9C2Zx4xzWykBhFAeYp9gkLyZE0zIMno9n4g19hmdY39scwqVkOXaxph0ZCXQcqxSpgQpONa59wkRDOL93eAXvimwlbPbwwVAegLS1HGfZAAAAABJRU5ErkJggg==\") no-repeat left center;background-size: 1em;}</style>";
const char HTTP_SCRIPT[] PROGMEM          = "<script>function c(l){document.getElementById('ssid').value=l.innerText||l.textContent;document.getElementById('pass').focus();}</script>";
const char HTTP_HEAD_END[] PROGMEM        = "</head><body><div style='text-align:left;display:inline-block;min-width:260px;' class='{c}'>";
const char HTTP_PORTAL_OPTIONS[] PROGMEM  = "<form action=\"/wifi\" method=\"get\"><button>Configure WiFi</button></form><br/><form action=\"/0wifi\" method=\"get\"><button>Configure WiFi (No Scan)</button></form><br/><form action=\"/i\" method=\"get\"><button>Info</button></form><br/><form action=\"/r\" method=\"post\"><button>Reset</button></form>";
const char HTTP_ITEM[] PROGMEM            = "<div><a href='#p' onclick='c(this)'>{v}</a>&nbsp;<span class='q {i}'>{r}%</span></div>";
const char HTTP_FORM_START[] PROGMEM      = "<form method='get' action='{a}'>";
const char HTTP_FORM_PARAM[] PROGMEM      = "<br/><input id='{i}' name='{n}' type='{t}' length='{l}' placeholder='{p}' value='{v}' {c}>{text}</input>";
const char HTTP_FORM_BUTTON[] PROGMEM     = "<br/><button type='{t}'>{text}</button>";
const char HTTP_A_HREF[] PROGMEM          = "<a id='{i}' name='{n}' href='{u}' {f}>{text}</a>";
const char HTTP_AJAX_BUTTON[] PROGMEM     = "<br/><button id='{i}' name='{n}' type='button' onClick='{f}'>{text}</button>";
const char HTTP_AJAX_SCRIPT[] PROGMEM     = "<script>function evalJSFromHtml(html) { var newElement = document.createElement('div'); newElement.innerHTML = html; var scripts = newElement.getElementsByTagName('script'); for (var i = 0; i < scripts.length; ++i) { var script = scripts[i]; eval(script.innerHTML); }} function {f}(o) { if(o){ url = o; } else { url = {u}; } var timeout = setInterval( function() { document.getElementById(\"{i}\").innerHTML = document.getElementById(\"{i}\").innerHTML + '<span style=\"font-size:5px\">a</span>'; }, 500); var xhttp = new XMLHttpRequest(); xhttp.onreadystatechange = function() { if (this.readyState == 4 && this.status == 200) { clearInterval(timeout); document.getElementById(\"{i}\").innerHTML = this.responseText; evalJSFromHtml(this.responseText); } }; xhttp.open(\"GET\", url, true); xhttp.timeout = 8000; xhttp.ontimeout = function() { clearInterval(timeout); document.getElementById(\"{i}\").innerHTML = \"Час очікування вийшов. Спробуйте ще раз.\";}; xhttp.send(); } </script>";

void setup() {
  Serial.begin(74880);
  
  logger("Startup timing: ");

}

void loop() {
}

void logger(String str) {
  Serial.println(str);
}
```

artua, 2018-06-22:
Hi!

Any updates?

artua, 2018-07-10:
@facchinm did you find the cause?

facchinm, 2018-07-11:
I just tested it again with latest IDE beta, ArduinoJson 6.1.0-beta and ESP8266 core v 2.4.0 and the MCVE compiles correctly now. We surely didn't fix `arduino-preprocessor` in the meantime so probably the fix was applied at core or library level. 
The cause is the preprocessor, since moving from `ctags` to `clang` has many advantages but also a lot of corner cases (see https://github.com/arduino/arduino-preprocessor/issues/6). 
Let me know if it also works on your setup, otherwise I'd move the issue to the preoprocessor repo, thanks!

per1234, 2018-09-26:
@artua have you had a chance to test the latest Arduino IDE beta build yet? If so, did it resolve the issue for you?

per1234, 2018-11-03:
It also works for me with Arduino IDE 1.9.0-beta build 87, even using the same ArduinoJson 5.13.1 version as @artua.

Since we haven't gotten any response and it's working for facchinm and me, I'll go ahead and close this. @artua if you are still having the problem with the latest beta build then please comment here and I'll reopen the issue.

closed 2018-11-03 by per1234

artua, 2018-11-06:
This sketch is working for me, but I tried another one with big files loaded via PROGMEM, it fails to compile with the same "'somefunction' was not declared in this scope" errors. I'll post MCVE if I can find the cause.

BTW, now 1.9-beta works better than current 1.8.7-release with these files.
Maybe you can advise me on any limitation to PROGMEM using ESP8266 platform? I can't find any info on that.

Thanks!