arduino/Arduino #7670: [Library Manager] New library - Arduino_SegmentDisplay_CD4511B

lecsDragos, 2018-06-05:
Hi there,

I have created a library and would like to share it with others via the  Library Manager. 
The code I wrote helps drive 7 segment displays using CD4511B chips.
Here is a link to it: [https://github.com/lecsDragos/Arduino_SegmentDisplay_CD4511B](https://github.com/lecsDragos/Arduino_SegmentDisplay_CD4511B)  

Please let me know if I need to make any changes in order for it to get published. 

Thanks,
Alex

per1234, 2018-06-05:
In order for your library to be added to the Library Manager index, it's required that you add a file named library.properties to the root folder of the repository. You can find the specifications for this file here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
After adding the file to your library, create or redo a release/tag and then comment here so we know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

lecsDragos, 2018-06-05:
added the required file and updated the tag

cmaglie, 2018-06-06:
library added

closed 2018-06-06 by cmaglie