arduino/Arduino #7671: Renamed Arduino Library on Github

jonseitz-garmin, 2018-06-06:
The previous library name was - 
https://github.com/garmin/LIDARLite_v3_Arduino_Library

The new library name is -
https://github.com/garmin/LIDARLite_Arduino_Library

I would like to change the name of the library in Arduino's Library Manager to that reflected in the repo and also add the 2.0.0 release to the install-able versions.

Thanks.

per1234, 2018-06-06:
Since GitHub redirects the old URL when you change the repository name, I think the primary difference that needs to be accommodated is the `name` value change from ["LIDAR-Lite v3"](https://github.com/garmin/LIDARLite_Arduino_Library/blob/1.0.3/library.properties#L1) to ["LIDAR-Lite"](https://github.com/garmin/LIDARLite_Arduino_Library/blob/2.0.0/library.properties#L1).

This will be done on cmaglie's next run through the Library Manager requests and your 2.0.0 release should be added at that time. This issue will be closed at that time and so you'll get a notification.

cmaglie, 2018-06-13:
done

closed 2018-06-13 by cmaglie