arduino/Arduino #7672: Please add my library for the multifunctional button into index

slavaza, 2018-06-07:
I want propose simple of use detector of the button pressing.  Add this library into index please: https://github.com/slavaza/SuperButton  Thank you vary match.

per1234, 2018-06-07:
There is already a file named "Thread" in the Library Manager index (in fact, it's your library). I believe that may have been a copy/paste artifact in your library.properties. Please change the `name` value in your library.properties to a unique `name` value and then either create a new release/tag or redo the existing tag.

Once that's done, comment here to let us know we can proceed.

slavaza, 2018-06-07:
Yes, it is my mistake. Excuse me, please. I hope that I fixed all.

per1234, 2018-06-07:
That will work, but in the future you should remember to update the `version` field in library.properties before creating a release.

cmaglie, 2018-06-13:
done

closed 2018-06-13 by cmaglie

slavaza, 2018-06-13:
Thank you very much.