arduino/Arduino #7676: subtract greater unsigned value from smaller unsigned = problem

mgx0, 2018-06-08:
Hi,

just hit this issue and I'm not sure whether this is expected or not. when performing calculation with unsigned values (substracting greater value from smaller), Arduino Returns max uint value (when using ULONG, returns max ulong). 

When doing the same with signed INTs, no problem here.

This happens when running this program on MEGA32u4 and does not happen when running the same program on ESP8266, PlatformIO and Arduino IDE used, results are the same. Didn't try MEGA328 though

I hit this problem when doing a fish feeder with millis() condition _if(abs(millis()-lastFeed)>feedFreq){ ... }_

can anybody explain what is going on here and why the behaviour of MEGA is different than on ESP8266?

```
void setup(){

Serial.begin(115200);
delay(5000);

}

void loop(){

 unsigned int a=4;
 unsigned int b=3;
 unsigned int c=2;
 unsigned int d=1;

  Serial.println("");
  Serial.println("abs(a-b)="+String(abs(a-b)));
  Serial.println("abs(b-c)="+String(abs(b-c)));
  Serial.println("abs(c-d)="+String(abs(c-d)));
  Serial.println("abs(d-a)="+String(abs(d-a)));

  Serial.println("----------");
  Serial.println("String(a-b)="+String(a-b));
  Serial.println("String(b-c)="+String(b-c));
  Serial.println("String(c-d)="+String(c-d));
  Serial.println("String(d-a)="+String(d-a));

  Serial.println("----------");
  Serial.print("a-b="); Serial.println(a-b);
  Serial.print("b-c="); Serial.println(b-c);
  Serial.print("c-d="); Serial.println(c-d);
  Serial.print("d-a="); Serial.println(d-a);

  Serial.println("----------");
  Serial.println("1-4="+String(1-4));

  Serial.println("----------"); 
  Serial.println("----------");
  Serial.println("----------");

delay(2000);

}
```

output: 

```
abs(a-b)=1
abs(b-c)=1
abs(c-d)=1
abs(d-a)=65533
----------
String(a-b)=1
String(b-c)=1
String(c-d)=1
String(d-a)=65533
----------
a-b=1
b-c=1
c-d=1
d-a=65533
----------
1-4=-3
----------
----------
----------

```

x0r19x91, 2018-06-08:
This is Unsigned Subtraction
`
1-4 = -3. Two's complement representation of -3 in 16 bits is 0x10000 - 3 = 0xfffd = 65533
`

closed 2018-06-08 by per1234

mgx0, 2018-06-08:
Okay.

Why esp8266 is returning d-a=-3 then? Mega32u4 returns 65533 as seen above

per1234, 2018-06-08:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to answer all your questions over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).