arduino/Arduino #7677: Error when platform is already installed (Arduino 1.8.5)

patrickelectric, 2018-06-08:
I think that the program should not terminate with a crash !
```
java.lang.Exception: Platform is already installed!
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:78)
	at processing.app.Base.<init>(Base.java:323)
	at processing.app.Base.main(Base.java:144)
make: *** [Makefile:27: deps] Error 255
```

matthijskooijman, 2018-06-09:
Can you provide a bit more details on what you did to obtain this error? Is it reproducible?

patrickelectric, 2018-06-09:
I run this command more than once.
```
arduino --pref boardsmanager.additional.urls="http://arduino.esp8266.com/stable/package_esp8266com_index.json" --save-prefs
arduino --install-boards esp8266:esp8266
```