arduino/Arduino #7682: Please add my library to the library manager for download

HullabalooRobotics, 2018-06-11:
https://github.com/HullabalooRobotics/Hyperduino-Library

per1234, 2018-06-11:
For inclusion in the Library Manager index, the library must contain a release that meets the requirements listed in the [FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ).

At the time of the libraries sole release, the library did not meet these requirements due to the lack of a library.properties file.

I see that since that release you have added a library.properties file but I identified a potential problem. There is a leading space in the `name` value, which is expressly prohibited by the Arduino Library Specification:

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
>- name - the name of the library. Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces (), underscores (_), dots (.) and dashes (-). _**It cannot start or end with a space**_,

I have submitted a pull request to fix that issue: https://github.com/HullabalooRobotics/Hyperduino-Library/pull/1

Once you have made a tag/release that contains a compliant version of the library please comment here so we know we can proceed.

HullabalooRobotics, 2018-06-11:
I believe I’ve made the necessary changes. Please let me know if this is acceptable.

HullabalooRobotics, 2018-06-13:
To preface this question, I'm new to this side of the Arduino process (writing/publishing a library). In light of that, is there a way to tell when my library will be officially integrated into the library manager?

per1234, 2018-06-13:
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of the issue closure and your library will be available in Library Manager shortly after that.

HullabalooRobotics, 2018-06-13:
Sounds good. Do you know how frequently this occurs? I'm attempting to set up the library ASAP for a computer-science related summer camp I'm helping with.

cmaglie, 2018-06-13:
done

closed 2018-06-13 by cmaglie