arduino/Arduino #7687: Add "IoTtweet" library to Library manager Arduino IDE

tsirinya, 2018-06-12:
We are developer team from http://www.iottweet.com which provide and IoT user disignable dashboard. It's compatible use with ESP8266 WiFi module, NodeMCU, WeMos, Nano32 and several IoT model.

github url : https://github.com/tsirinya/IoTtweet

Thank you very much
IoTtweet team.

per1234, 2018-06-12:
The repository doesn't contain a library, only a library.properties file and a readme.

Once you have added a library and created a new release/tag please comment here and we'll go ahead and add it to the Library Manager index.

CornyjK, 2018-06-17:
I would reccomend to rename mylibrary.h/cpp to something better (like IoTtweet.h/cpp). Ideally move it do "/src" direktory. IMO your tag version should match with library version. Not sure what "Arduino libraries" file does (delete it?).

per1234, 2018-06-17:
It looks like there's already a well-formed IoTtweet library here:
https://github.com/Isaranu/IoTtweet
It was added to the Library Manager index some time ago (https://github.com/arduino/Arduino/issues/5704).

closed 2018-06-17 by per1234