arduino/Arduino #7689: ArduinoToBreadboard Tutorial

jt530, 2018-06-13:
Hi guys,

I think there is an error on the tutorial on the page: https://www.arduino.cc/en/Tutorial/ArduinoToBreadboard

To burn a bootloader to an ATMEGA328P-PU using an Arduino Uno, the tutorial suggests setting the board in the Arduino IDE to "Arduino Duemilanove or Nano w/ ATmega328" from the Tools > Board menu. (Or "ATmega328 on a breadboard (8 MHz internal clock)".  However, trying as hard as I can, this will not work.  Leaving the board set as an Arduino Uno works perfectly though.  

Could this please be changed because I'm sure it confuses a lot of people.

Thanks,
Jason

per1234, 2018-06-18:
The only difference between the Arduino Duemilanove or Diecimila (ATmega328P) and the Uno boards definitions is the bootloader, boot section size, and the upload speed (because of the different bootloader). That one works fine for me so I think whatever problem you're having with the Duemilanove is not universal.

It does seem reasonable to change the instructions to the Uno since that provides a smaller bootloader. I wonder if the faster upload speed of the Uno vs Duemilanove could be less reliable on a breadboarded circuit? I haven't checked what the baud error is on 57600 vs 115200.

As for that "breadboard" package, I also found that to be broken and I have seen at least one report on the Arduino forum of that causing a lot of confusion for someone. I've been meaning to investigate that further and report the issue.

jt530, 2018-06-19:
I don't know why I can't get the Duemilanove profile to work unfortunately.  It seems sensible though to change it to the Uno based on what you've said.  I'm not sure what you mean when you are referring to the 'breadboard package' though.

Best regards,
Jason