arduino/Arduino #7692: Please add our OLED library

durydevelop, 2018-06-14:
Please add our library:
https://github.com/durydevelop/arduino-lib-oled.git

Thanks.

per1234, 2018-06-14:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

durydevelop, 2018-06-14:
ok now match, sorry

per1234, 2018-06-14:
There are still no tags or releases:
https://github.com/durydevelop/arduino-lib-oled/tags

Did you maybe forget to push it to GitHub?

durydevelop, 2018-06-15:
Sorry, I'm newbie on it, I missed to explicit push the tag.
So, now may be ok.

cmaglie, 2018-06-21:
done

closed 2018-06-21 by cmaglie