arduino/Arduino #7702: [Library Manager] Please add blynk-library-for-chinese library

hznupeter, 2018-06-18:
https://github.com/hznupeter/blynk-library-for-chinese

i translate the blynk library into chinese,i hope chinese user can ues the lib freely

per1234, 2018-06-18:
Before we can add your library to the Library Manager index, you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

In this case there may be an issue in that, according to [semver](https://semver.org/), your version ([0.5.3-2](https://github.com/hznupeter/blynk-library-for-chinese/blob/master/library.properties#L2)) is considered to be lower than the last release version ([0.5.3](https://github.com/hznupeter/blynk-library-for-chinese/releases/tag/v0.5.3)). This is because the `-2` part is the pre-release syntax. I'm not sure exactly how the Library Manager indexer handles that. I'm sure cmaglie can provide an answer when they next go through the Library Manager inclusion requests.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

hznupeter, 2018-06-18:
@per1234  thks,i have fix it.

per1234, 2018-06-19:
@hznupeter, did you close the issue because you no longer wish your library to be added to the Library Manager index?

hznupeter, 2018-06-20:
@per1234  sorry,i wish my library will be added.
I thought the issue should be closed after you know it.
So,i will keep it open.

per1234, 2018-06-20:
cmaglie will close this issue once your library has been added.

njh, 2019-10-27:
@hznupeter looks like you have now deleted the library.
Any information about why?