arduino/Arduino #7706: Mistake on website

Rdgraaff, 2018-06-18:
Dear Arduino,

I've spotted a mistake on your website, (https://www.arduino.cc/en/Tutorial/Genuino101CurieIMUOrientationVisualiser).
In the second sentence (While the **accelerometer** is able to determine the orientation of the board, the **gyroscope** measures the angular velocity of th
 accelerometer is able to determine the orientation of the board, the gyroscope measures the angular velocity of the board). You wrote accelerometer in stead of gyroscope, where gyroscope should have stood, and the other way around.

matthijskooijman, 2018-06-18:
I don't actually think that is wrong. An accelerometer can determine the orientation of the board (relative to gravity), though it can *also* determine acceleration of course. A gyroscope measures angular velocity, iow how fast the board is rotating and along which axis.

Rdgraaff, 2018-06-18:
Oh oke bedankt! Didnt know that yet, and btw are you dutch, because I got a
problem( trying to figure out how to move a DC motor and a servo at the
same time) and its a little easier for me to talk dutch.

Kind regards/groetjes,
Ruben

closed 2018-06-19 by per1234