arduino/Arduino #7709: Add new version of bossac

frippe75, 2018-06-20:
As of https://github.com/shumatech/BOSSA/issues/33
 
There is a way to allow larger bootloaders than the default 8k (-e option of bossac).
Would be nice to see this integrated into the IDE as well.

Maybe add a new property in the boards.txt that gets picked up and passed to bossac.

frippe75, 2018-06-20:
Just realize there are bossac 1.8.0 in other directories. Need to workout how to call them

facchinm, 2018-06-25:
Hi @frippe75 ,
to target the new bossac you need to:
1- "link" it to your platform by adding 
```
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "bossac",
              "version": "1.8.0-48-gb176eee"
            }
```
in `package_$yourboard_index.json`

2- replace all occurrences of `runtime.tool.bossac` in `platform.txt` with `runtime.tool.bossac-1.8.0-48-gb176eee` to make sure it gets picked up correctly.

Otherwise, you can add Adafruit's SAMD core json (https://adafruit.github.io/arduino-board-index/package_adafruit_index.json) that already requires it, install the core and then apply point 2)

closed 2018-06-25 by facchinm