arduino/Arduino #7710: Rename webthing-esp8266 to webthing-arduino

hobinjk, 2018-06-20:
We added support for WiFi101 and ESP32 boards to our webthing-esp8266 library (added in #7562) and would like to rename the library to webthing-arduino to reflect this increase in platforms.

The updated location of the library is https://github.com/mozilla-iot/webthing-arduino.

Thanks!

cmaglie, 2018-06-21:
I see that you updated the name in the `library.properites` but you didn't made a new release: https://github.com/mozilla-iot/webthing-arduino/blob/v0.2.0/library.properties I guess you did the name change after releasing the 0.2.0.

Before changing the name you should publish another release:
- edit `library.properties` and bump version
- commit & push
- create a new tag

hobinjk, 2018-06-21:
Thanks for the response! I've tagged the v0.3.0 release for the latest changes to library.properties. https://github.com/mozilla-iot/webthing-arduino/releases/tag/v0.3.0

cmaglie, 2018-06-21:
ok done!

closed 2018-06-21 by cmaglie