arduino/Arduino #7718: Comparing an unsigned it fails if it is at 0xFFFF.

cartolino, 2018-06-22:
Code snippets:

unsigned int toff; // also tried with uint16_t , same results
void setup(){ // .... .... .... .... 
1)  
  toff=EEPROM.read(Toff)+256*EEPROM.read(Toff+1);if ((toff)>30000) toff=1500;

2)
  toff=EEPROM.read(Toff)+256*EEPROM.read(Toff+1);if ((toff-1)>30000) toff=1500;

The idea is to put a sensible default in RAM if the EEPROM was not programmed.

Problem: after the 1st version of the code, toff==65535==0xFFFF ; Apparently the comparison fails with FFFF.

(The 2nd variant with the -1 hack was contributed by a colleague whom I asked to debug this.)

Checking the assembly code with avr-objdump shows only one difference in the 2 codes, the line shown with >>>>.

    1898:       90 93 d6 02     sts     0x02D6, r25     ; 0x8002d6 <toff+0x1>
    189c:       80 93 d5 02     sts     0x02D5, r24     ; 0x8002d5 <toff>
>>>>    18a0:       01 97           sbiw    r24, 0x01       ; 1
    18a2:       81 33           cpi     r24, 0x31       ; 49
    18a4:       95 47           sbci    r25, 0x75       ; 117
    18a6:       30 f0           brcs    .+12            ; 0x18b4 <main+0x340>
    18a8:       8c e4           ldi     r24, 0x4C       ; 76
    18aa:       94 e0           ldi     r25, 0x04       ; 4
    18ac:       90 93 d6 02     sts     0x02D6, r25     ; 0x8002d6 <toff+0x1>
    18b0:       80 93 d5 02     sts     0x02D5, r24     ; 0x8002d5 <toff>

Not sure why the comparison fails when r24==0xFF and r25==0xFF but passes when r24=0xFE.
And if it is an AVR issue (the MCU used was an Atmega324A with the MightyCore library from MCUDUDE, using LTO), then the compiler should probably adjust for this. 

Or is it an optimization issue?

At any rate, I fixed my project now with the hack, but it is still scary that one cannot compare for some input values.

If this is deemed a non-issue (due to it happening on 324A) is there any official way to prevent it other than the -1 which could fail on a zero?

cartolino, 2018-06-22:
More info: it seems that without the LTO optimization it compiles differently:
     cf2:       21 33           cpi     r18, 0x31       ; 49
     cf4:       35 47           sbci    r19, 0x75       ; 117
     cf6:       10 f0           brcs    .+4             ; 0xcfc <setup+0x27a>

so it could be an optimization issue after all....

Moral of the story: remove LTO optimization.

cartolino, 2018-06-22:
Tried without the LTO optimization, assembles differently but the same bug manifests itself :-( 

so not an optimization issue.

darrylp1per, 2018-06-28:
why not tell the compiler to use unsigned arithmetic when adding the two reads of the EEPROM ?

`ie.`   toff=(uint16_t)( EEPROM.read(Toff)+256*EEPROM.read(Toff+1) );  if ((toff)>30000) toff=1500;`

cartolino, 2018-06-28:
Hi Matthis,

Yes, 1) fails and 2)succeeds (0 is not valid either so if it overflows it's correct if it defaults to 1500).

I was trying to repro this issue on a mainstream CPU (Mega 2560), failed (I will keep trying), but found **another** interesting one:

```
unsigned int ui1;
uint16_t ui16;

#include <EEPROM.h>

#define sp Serial.print
#define spf(x) sp(F(x))
#define spnl spf("\r\n")

void setup() {
  Serial.begin(115200);while(!Serial){1;}delay(500);
  ui1=EEPROM.read(1023)+256*EEPROM.read(1023);
  ui16=EEPROM.read(1023)+256*EEPROM.read(1023);
  spf("Initial: ui1=");sp(ui1);spf("   ui16=");sp(ui16);spnl;
  //spf("sizeof(ui1)=");sp(sizeof(ui1));spf("   sizeof(ui16)=");sp(sizeof(ui16));spnl; //commenting this out makes ui1 32 bits long

  if (ui1>30000) ui1=111;
  if (ui16>30000) ui16=222;
  spf("After: ui1=");sp(ui1);spf("   ui16=");sp(ui16);spnl;
}

void loop() {  
 delay(1000);
}
```

This prints:

Initial: ui1=4294967295   ui16=65535
After: ui1=111   ui16=222

If I remove the comment to the line that prints the sizes, then it prints correctly:

Initial: ui1=65535   ui16=65535
sizeof(ui1)=2   sizeof(ui16)=2
After: ui1=111   ui16=222

Quantum entanglement :-) apparently.

Same happens if I swap the 2 lines (even though at least one pass of the compiler starts from the bottom):

sizeof(ui1)=2   sizeof(ui16)=2
Initial: ui1=65535   ui16=65535
After: ui1=111   ui16=222

My versions: Arduino 1.8.3, Arduino AVR boards 1.6.207.
Re-tried with the MiniCore 2.0.1 on Atmega 328pb, same results.
Same with MegaCore 2.0.0 Atmega 324A (this is where I found the original issue).

cartolino, 2018-06-29:
I managed to repro the ORIGINAL bug with the following file (removed most other stuff from it).
Note that variable names and comments ARE copyrighted by my employer, by contract, so don't copy+paste into code that's distributed somewhere else.

File is here: [bug-hunting2-324A-MightyCore2.0.0.txt](https://github.com/arduino/Arduino/files/2150338/bug-hunting2-324A-MightyCore2.0.0.txt)

Happy bug hunting!

The same error happens on 324A (with MightyCore 2.0.0 and LTO enabled) as well as on a standard Mega2560 with Arduino 1.8.3, Arduino AVR boards 1.6.207.

In the mean time, please let me know if you have any standard fix that can be proactively applied.
(This whole "don't divide, Intel inside" vibe erodes confidence in the embedded code generated by avr-gcc.)

`````
#include <EEPROM.h>

//#define DEMO 261
#define Toff 264
#define Ton  266
#define Tp0  268
#define Tp1  270
#define Tp2  272
#define Tp3  274

//unsigned long ms=0; int i;

#define sp Serial.print 
#define spf(x) sp(F(x))
#define spnl spf("\r\n")

//RAM=2048 bytes on this ATMega324A MCU ; 32K FLH and 1K EEPROM
//byte data[64];
//byte demo=0;

uint16_t toff; unsigned int ton,tp0,tp1,tp2,tp3;
void setup() {
  Serial.begin(38400);while(!Serial) {1;}  
  toff=EEPROM.read(Toff)+256*EEPROM.read(Toff+1);if ((toff)>0x7530) toff=1500;
  ton=EEPROM.read(Ton)+256*EEPROM.read(Ton+1);   if ((ton)>30000U) ton=2100;
  tp1=EEPROM.read(Tp1)+256*EEPROM.read(Tp1+1);   if ((tp1-1)>30000U) tp1=2100;
  tp2=EEPROM.read(Tp2)+256*EEPROM.read(Tp2+1);   if ((tp2-1)>30000U) tp2=2100;
  tp3=EEPROM.read(Tp3)+256*EEPROM.read(Tp3+1);   if ((tp3-1)>30000U) tp3=2100;
  tp0=EEPROM.read(Tp0)+256*EEPROM.read(Tp0+1);   if ((tp0-1)>30000U) tp0=2100;

  delay(500);

  spf("After the checks:  toff ");sp(toff);spf("  ton ");sp(ton);spf("   tp0 ");sp(tp0);
  spf("   tp1 ");sp(tp1);  spf("  tp2 ");sp(tp2);spf("  tp3 ");sp(tp3);spnl;

 
}

void loop() {  
}

```

cartolino, 2018-06-29:
... in the mean time found that if I use   if ( (toff-30000)>0) ... works properly for 0xFFFF .
Guess I'm gonna have to use this for 2-byte integers that can go to 0xFFFF from now on...

cartolino, 2018-06-29:
Tried to see what happens with the assembled code when using if ((x-limit)>0) versus y>limit:
```

  toff=EEPROM.read(Toff)+256*EEPROM.read(Toff+1);if ((toff- 30000 )>0) toff=1500;
  ton=EEPROM.read(Ton)+256*EEPROM.read(Ton+1);   if ((unsigned)ton>30000U) ton=2100;
```

Assembler looks like this:

```
  toff=EEPROM.read(Toff)+256*EEPROM.read(Toff+1);if ((toff- 30000 )>0) toff=1500;
 79e:   88 e0           ldi     r24, 0x08       ; 8
 7a0:   91 e0           ldi     r25, 0x01       ; 1
 7a2:   ae de           rcall   .-676           ; 0x500 <_ZN11EEPROMClass4readEi.isra.1>
 7a4:   f8 2e           mov     r15, r24
 7a6:   89 e0           ldi     r24, 0x09       ; 9
 7a8:   91 e0           ldi     r25, 0x01       ; 1
 7aa:   aa de           rcall   .-684           ; 0x500 <_ZN11EEPROMClass4readEi.isra.1>
 7ac:   90 e0           ldi     r25, 0x00       ; 0
 7ae:   98 2f           mov     r25, r24
 7b0:   88 27           eor     r24, r24
 7b2:   8f 0d           add     r24, r15
 7b4:   91 1d           adc     r25, r1
 7b6:   80 33           cpi     r24, 0x30       ; 48
 7b8:   25 e7           ldi     r18, 0x75       ; 117
 7ba:   92 07           cpc     r25, r18
 7bc:   11 f0           breq    .+4             ; 0x7c2 <main+0x16e>
 7be:   8c ed           ldi     r24, 0xDC       ; 220
 7c0:   95 e0           ldi     r25, 0x05       ; 5
 7c2:   90 93 26 02     sts     0x0226, r25     ; 0x800226 <toff+0x1>
 7c6:   80 93 25 02     sts     0x0225, r24     ; 0x800225 <toff>
  ton=EEPROM.read(Ton)+256*EEPROM.read(Ton+1);   if ((unsigned)ton>30000U) ton=2100;
 7ca:   8a e0           ldi     r24, 0x0A       ; 10
 7cc:   91 e0           ldi     r25, 0x01       ; 1
 7ce:   98 de           rcall   .-720           ; 0x500 <_ZN11EEPROMClass4readEi.isra.1>
 7d0:   f8 2e           mov     r15, r24
 7d2:   8b e0           ldi     r24, 0x0B       ; 11
 7d4:   91 e0           ldi     r25, 0x01       ; 1
 7d6:   94 de           rcall   .-728           ; 0x500 <_ZN11EEPROMClass4readEi.isra.1>
 7d8:   90 e0           ldi     r25, 0x00       ; 0
 7da:   98 2f           mov     r25, r24
 7dc:   88 27           eor     r24, r24
 7de:   8f 0d           add     r24, r15
 7e0:   91 1d           adc     r25, r1
 7e2:   90 93 24 02     sts     0x0224, r25     ; 0x800224 <ton+0x1>
 7e6:   80 93 23 02     sts     0x0223, r24     ; 0x800223 <ton>
 7ea:   81 33           cpi     r24, 0x31       ; 49
 7ec:   95 47           sbci    r25, 0x75       ; 117
 7ee:   34 f0           brlt    .+12            ; 0x7fc <main+0x1a8>
 7f0:   84 e3           ldi     r24, 0x34       ; 52
 7f2:   98 e0           ldi     r25, 0x08       ; 8
 7f4:   90 93 24 02     sts     0x0224, r25     ; 0x800224 <ton+0x1>
 7f8:   80 93 23 02     sts     0x0223, r24     ; 0x800223 <ton>

```

cartolino, 2018-06-29:
In http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-0856-AVR-Instruction-Set-Manual.pdf , on page 21, ch2.3 , it mentions that "BRLT" is for SIGNED comparisons, so the compiler is compiling incorrect code here. 
It should have been "BRLO".

cartolino, 2018-07-04:
So, is this a GCC bug or an Arduino tool chain bug?
(Meaning, who needs to be notified in order to get it fixed?)

cartolino, 2018-07-04:
For MiniCore and MegaCore, I used the LTO option so it may be that the code above was using that. (Both had the same result, so either it's not an LTO issue or they are both broken.)

My GCC version is "tools/avr-gcc/5.4.0-atmel3.6.1-arduino2" - 

```
$ /c/progra~2/Atmel/Studio/7.0/toolchain/avr8/avr8-gnu-toolchain/bin/avr-g++ -v
Using built-in specs.
Reading specs from c:/progra~2/atmel/studio/7.0/toolchain/avr8/avr8-gnu-toolchain/bin/../lib/gcc/avr/5.4.0/device-specs/specs-avr2
COLLECT_GCC=C:\progra~2\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-g++.exe
COLLECT_LTO_WRAPPER=c:/progra~2/atmel/studio/7.0/toolchain/avr8/avr8-gnu-toolchain/bin/../libexec/gcc/avr/5.4.0/lto-wrapper.exe
Target: avr
Configured with: /home/toolsbuild/workspace/avr8-gnu-toolchain/src/gcc/configure LDFLAGS=-L/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-win32_x86-hostlibs/lib CPPFLAGS= --target=avr --host=i686-w64-mingw32 --build=x86_64-pc-linux-gnu --prefix=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-win32_x86 --libdir=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-win32_x86/lib --enable-languages=c,c++ --with-dwarf2 --enable-doc --disable-shared --disable-libada --disable-libssp --disable-nls --with-avrlibc=yes --with-mpfr=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-win32_x86-hostlibs --with-gmp=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-win32_x86-hostlibs --with-mpc=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-win32_x86-hostlibs --enable-win32-registry=avrtoolchain --with-pkgversion=AVR_8_bit_GNU_Toolchain_3.6.1_1750 --with-bugurl=http://www.atmel.com
Thread model: single
gcc version 5.4.0 (AVR_8_bit_GNU_Toolchain_3.6.1_1750)

```

cartolino, 2018-07-04:
oops, pressed the close button by mistake. 
The issue is still open.

darrylp1per, 2018-07-05:
my 4.8.5 has LTO enabled just for info.

`D:\AVR\arduino-1.5.5\hardware\tools\avr\bin>avr-g++ -v
Using built-in specs.
COLLECT_GCC=avr-g++
COLLECT_LTO_WRAPPER=d:/avr/arduino-1.5.5/hardware/tools/avr/bin/../libexec/gcc/avr/4.8.5/lto-wrapper.exe
Target: avr
Configured with: ../configure --prefix=/usr/local/avr-4.8.5/avr-win --target=avr --enable-long-long --enable-fixed-point --enable-languages=c,c++ --enable-lto --disable-checking --disable-nls --disable-doc --disable-libssp --disable-libada --disable-shared --with-avrlibc=yes --with-dwarf2 --host=i686-w64-mingw32 --build=i686-pc-linux-gnu
Thread model: single
gcc version 4.8.5 (GCC)`

cartolino, 2018-07-09:
ok, so if I understand this correctly, the bug has been fixed in 8.1.0 but 8.1.0 has not been yet adopted by Arduino.
In this case what are the options? 
* Manually installing this new gcc (how? what else can it impact?)      
* someone in the Arduino team would integrate this properly and then notify us here?
* same but no notification, we need to check some page for progress?