arduino/Arduino #7721: Accesibility is Denied

kaloliaalap, 2018-06-22:
Hello, 

When I tried to access https://www.arduino.cc/reference/en/language/functions/main/setup/ 
from the page 
https://www.arduino.cc/en/Tutorial/Blink
,  

It says access is denied. 

Sorry, If it's been already resolved and I couldn't find it.

per1234, 2018-06-23:
Thanks for reporting this @kaloliaalap!

The correct link should be:
https://www.arduino.cc/reference/en/language/structure/sketch/setup/

The same issue also occurs on that page for the link to the `loop()` reference page:
https://www.arduino.cc/en/Reference/Loop
which redirects to:
https://www.arduino.cc/reference/en/language/functions/main/loop/
but instead should redirect to:
https://www.arduino.cc/reference/en/language/structure/sketch/loop/

per1234, 2018-06-23:
We need to leave this open until it's fixed.

gabru-md, 2018-06-27:
Is someone working on this issue? :sweat_smile:

per1234, 2020-05-23:
All redirects for links on the Blink tutorial page are now working correctly for me so I'm closing this.

closed 2020-05-23 by per1234