arduino/Arduino #7723: Arduino Functions could return An Array Variable?

IngenieroDFZ, 2018-06-23:
I have some troubles to compilate the follow code, why?
If you have programmed in JAVA anytime, you know that this kind of code is possible to operate in JAVA, why is not possible in Arduino IDE?

So please Pay attenction: 

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
  int B[5]={0,0,0,0,0};
  int C=0;
  C=funcionX(); //This  ocde line is ok
  B=funcionY(); // but this code line is not ok.
}

int funcionX() //this function is ok
{
  return 3;
}

int[5] funcionY() // But this function is not ok
{
  int A[5]={1,22,333,4444,55555};
  return A;
}

How can I make functions with  Array  return  in Arduino?

IngenieroDFZ, 2018-06-23:
???

PaulStoffregen, 2018-06-23:
You're asking in the wrong place.  This issue tracker is meant for reporting defects with Arduino's software.  It is *not* the proper place to ask for help.  The forum is where you should ask this question.

https://forum.arduino.cc/

closed 2018-06-23 by per1234