arduino/Arduino #7724: Mac OS 10.13.5: Serial Monitor shows "???" instead of ascii

surge603, 2018-06-24:
Mac OS 10.13.5: Arduino IDE 1.8.5 Serial Monitor shows "???" instead of ascii

per1234, 2018-06-24:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2018-06-24 by per1234

PaulStoffregen, 2018-06-24:
If this really is a defect in the Arduino software, please show us a screenshot of the problem and a copy of the program you ran on your Arduino board.