arduino/Arduino #7730: MacOS 'close window' terminates application

a7hybnj2, 2018-06-26:
Current behavior: When you close the open windows the application terminates.

Desired behavior: When you close the open windows keep the application running as just a menu. It would still populate the 'dock' and you could still alt+tab to focus it. Then you could open or new another instance.

Many IDE's, text editors, and other applications have my 'desired behavior': mvim, sublime, atom, kicad, steam, ect...

marcelstoer, 2018-09-12:
Not sure if this is the standard behavior with Java desktop applications on macOS but it's definitely annoying.

kosso, 2018-11-02:
This sort of thing is certainly optional when it comes to apps built with Electron.  Java should also have that option.

facchinm, 2018-11-05:
https://github.com/arduino/Arduino/pull/8023 should solve the issue :slightly_smiling_face:

a7hybnj2, 2018-12-06:
I am going to close this... version 1.9 has this functionality which is totally awesome! Thanks!

closed 2018-12-06 by a7hybnj2