arduino/Arduino #7731: Editing values in ~/.arduino/preferences.txt does absolutely nothing.

odellus, 2018-06-26:
Yeah I'm trying to add a C++ compiler flag, so I add 
```
compiler.cpp.flags=-std=c++11
```
and it has no effect. I'm using default ubuntu install right now, but I'm opening an issue already bc this is just ridiculous. Nothing I'm doing is putting the flag in there.

When I run:

```
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -std=c++11 -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega2560 -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/cores/arduino -I/usr/share/arduino/hardware/arduino/variants/mega -I/usr/share/arduino/libraries/Wire -I/usr/share/arduino/libraries/LiquidCrystal -I/usr/share/arduino/libraries/SPI /tmp/build5641016476884757357.tmp/Marlin.cpp -o /tmp/build5641016476884757357.tmp/Marlin.cpp.o
```
which is the command + -std=c++11 from the terminal, it works. So I'm looking into using avrdude now to upload the compiled marlin firmware to my ATMega board and it's a huge PITA.

Thanks!

matthijskooijman, 2018-06-26:
Huh? The compiler flags are not defined in preferences.txt, but in `platform.txt` for the platform you are using (which is either stored in the Arduino installation directory, for AVR in older versions, or in your homedir for newer [possibly not yet released] versions or non-AVR cores).

Also, `~/.arduino` is for Arduino 1.0.x, since 1.5.0 stuff is stored in `~/.arduino15`.

If this solves your issue, do you have any suggestion on where documentation could have been better to have prevented your issue?

Finally: This is not really the right place for questions, but only for (well-documented) bug reports and feature requests (for future reference).

matthijskooijman, 2018-06-26:
One more thing: C++11 has been enabled in the AVR core since IDE version 1.6.something IIRC (it's certainly enabled in the latest versions), so I suspect you're running an older version? Upgrading might be the easiest approach.

odellus, 2018-06-26:
Fresh install on Ubuntu

odellus, 2018-06-26:
I installed arduino with `sudo apt install arduino`
Is this not a supported method of installing? Or just not recommended?

This is where arduino installs on Ubuntu.
```
thomas@machine-001:/usr/share/arduino$ ls
arduinopc.jar  hardware  libraries  revisions.txt
examples       lib       reference  tools
thomas@machine-001:/usr/share/arduino$ find . | grep platform
thomas@machine-001:/usr/share/arduino$
```
There are no platform.txt files in here to edit.

odellus, 2018-06-26:
It says Arduino 2:1.0.5 in the title of the Arduino GUI.

Is the apt version really 1.0.5? I thought it was 2.something. Oh my goodness.

https://playground.arduino.cc/Linux/Debian

Closing. SMH

closed 2018-06-26 by odellus

matthijskooijman, 2018-06-26:
The 2: prefix is an epoch, used to ensure incremental ordering when Arduino changed their version scheme years ago. You're not the first one to fall for that :-)

The Ubuntu/Debian version has not been updated for a long time. At the time, there were some licensing unclarities, which have not been seeing a lot of progress (not sure what the next steps are there, tbh...).

In any case, good luck with a newer version, and thanks for reporting back!