arduino/Arduino #7743: Unable to Paste file name by keyboard shortcut in Save As dialog

ddownn, 2018-06-29:
Keyboard shortcuts for Select All (Cmd-A) and Paste (Cmd-V) are not working when the cursor is in the file name text input of the Save As dialog box. Right clicking and using the context menu in the file name input does work to Select All and Paste.

Arduino 1.8.5
Mac OS 10.13.4

per1234, 2018-06-29:
This was previously reported in https://github.com/arduino/Arduino/issues/1949 but closed due to lack of response. I can't reproduce it on the Windows or ARM Linux versions of Arduino IDE 1.8.5. The previous issue report mentioned the "Apple" key so they may have also been using Mac.

ddownn, 2018-07-01:
I don’t remember if it worked on a previous version of the IDE. 

The system is a MacBook Pro using the built-in keyboard. 

I also tried a Logitech MK700 via USB, and an Omoton Bluetooth keyboard and these also do not paste. 

The Edit menu item does flash when the shortcut is typed, and pasting works fine in the code editor.

ddownn, 2018-07-22:
For what it's worth, although this may be completely unrelated, I've just come to realize that Processing 3.3.7 behaves the same way.