arduino/Arduino #7747: 1.9.0 beta   SET_OUTPUT(SPI_SCK_PIN);

dmtulsa, 2018-07-01:
1.8.5 does not produce this error

Error:
Arduino: 1.9.0-beta (Windows 10), Board: "Sanguino, ATmega1284 or ATmega1284P (16 MHz)"

In file included from sketch\MarlinConfig.h:26:0,

                 from sketch\Sd2Card.cpp:29:

Sd2Card.h:100:38: error: pasting "/* SPI Master In Slave Out pin*/" and "_DDR" does not give a valid preprocessing token

  #define SPI_MISO_PIN MISO_PIN       // SPI Master In Slave Out pin

                                      ^

sketch\fastio.h:94:35: note: in definition of macro '_SET_INPUT'

 #define _SET_INPUT(IO) do {DIO ## IO ## _DDR &= ~_BV(DIO ## IO ## _PIN); } while (0)

                                   ^

sketch\Sd2Card.cpp:316:3: note: in expansion of macro 'SET_INPUT'

   SET_INPUT(SPI_MISO_PIN);

   ^

sketch\Sd2Card.cpp:316:13: note: in expansion of macro 'SPI_MISO_PIN'

   SET_INPUT(SPI_MISO_PIN);

             ^

Sd2Card.h:100:38: error: pasting "/* SPI Master In Slave Out pin*/" and "_PIN" does not give a valid preprocessing token

  #define SPI_MISO_PIN MISO_PIN       // SPI Master In Slave Out pin

                                      ^

sketch\fastio.h:94:61: note: in definition of macro '_SET_INPUT'

 #define _SET_INPUT(IO) do {DIO ## IO ## _DDR &= ~_BV(DIO ## IO ## _PIN); } while (0)

                                                             ^

sketch\Sd2Card.cpp:316:3: note: in expansion of macro 'SET_INPUT'

   SET_INPUT(SPI_MISO_PIN);

   ^

sketch\Sd2Card.cpp:316:13: note: in expansion of macro 'SPI_MISO_PIN'

   SET_INPUT(SPI_MISO_PIN);

             ^

Sd2Card.h:99:39: error: pasting "/* SPI Master Out Slave In pin*/" and "_DDR" does not give a valid preprocessing token

   #define SPI_MOSI_PIN MOSI_PIN       // SPI Master Out Slave In pin

                                       ^

sketch\fastio.h:95:36: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO) do {DIO ## IO ## _DDR |= _BV(DIO ## IO ## _PIN); } while (0)

                                    ^

sketch\Sd2Card.cpp:317:3: note: in expansion of macro 'SET_OUTPUT'

   SET_OUTPUT(SPI_MOSI_PIN);

   ^

sketch\Sd2Card.cpp:317:14: note: in expansion of macro 'SPI_MOSI_PIN'

   SET_OUTPUT(SPI_MOSI_PIN);

              ^

Sd2Card.h:99:39: error: pasting "/* SPI Master Out Slave In pin*/" and "_PIN" does not give a valid preprocessing token

   #define SPI_MOSI_PIN MOSI_PIN       // SPI Master Out Slave In pin

                                       ^

sketch\fastio.h:95:61: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO) do {DIO ## IO ## _DDR |= _BV(DIO ## IO ## _PIN); } while (0)

                                                             ^

sketch\Sd2Card.cpp:317:3: note: in expansion of macro 'SET_OUTPUT'

   SET_OUTPUT(SPI_MOSI_PIN);

   ^

sketch\Sd2Card.cpp:317:14: note: in expansion of macro 'SPI_MOSI_PIN'

   SET_OUTPUT(SPI_MOSI_PIN);

              ^

Sd2Card.h:104:39: error: pasting "/* SPI Clock pin*/" and "_DDR" does not give a valid preprocessing token

   #define SPI_SCK_PIN SCK_PIN         // SPI Clock pin

                                       ^

sketch\fastio.h:95:36: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO) do {DIO ## IO ## _DDR |= _BV(DIO ## IO ## _PIN); } while (0)

                                    ^

sketch\Sd2Card.cpp:318:3: note: in expansion of macro 'SET_OUTPUT'

   SET_OUTPUT(SPI_SCK_PIN);

   ^

sketch\Sd2Card.cpp:318:14: note: in expansion of macro 'SPI_SCK_PIN'

   SET_OUTPUT(SPI_SCK_PIN);

              ^

Sd2Card.h:104:39: error: pasting "/* SPI Clock pin*/" and "_PIN" does not give a valid preprocessing token

   #define SPI_SCK_PIN SCK_PIN         // SPI Clock pin

                                       ^

sketch\fastio.h:95:61: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO) do {DIO ## IO ## _DDR |= _BV(DIO ## IO ## _PIN); } while (0)

                                                             ^

sketch\Sd2Card.cpp:318:3: note: in expansion of macro 'SET_OUTPUT'

   SET_OUTPUT(SPI_SCK_PIN);

   ^

sketch\Sd2Card.cpp:318:14: note: in expansion of macro 'SPI_SCK_PIN'

   SET_OUTPUT(SPI_SCK_PIN);

              ^

exit status 1
pasting "/* SPI Master In Slave Out pin*/" and "_DDR" does not give a valid preprocessing token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

per1234, 2018-07-01:
I believe this is https://github.com/arduino/arduino-preprocessor/issues/6

Also related: https://github.com/arduino/Arduino/issues/5079#issuecomment-368367321

facchinm, 2018-07-16:
It is, indeed. Closing as duplicate, please continue the discussion on #5079

closed 2018-07-16 by facchinm