arduino/Arduino #7751: OSX MacOS 10.13.5 and IDE 1.8.5 long startup time

downunder54, 2018-07-04:
Hello again,

I also noticed that the initial window saying "Initializing packages..." is sitting there for about 10 sec before proceeding to load the rest. The previous version loads straight away.

On the console I get:

Could not find boards.txt in /Users/hze/Documents/Arduino/hardware/LeoStick/bootloaders. Is it pre-1.5?
Could not find boards.txt in /Users/hze/Documents/Arduino/hardware/LeoStick/cores. Is it pre-1.5?
WARNING: Error loading hardware folder /Users/hze/Documents/Arduino/hardware/LeoStick
  No valid hardware definitions found in folder LeoStick.

right before this delay, nothing more afterwards.

per1234, 2018-07-04:
Regarding the slower startup. That was already reported in https://github.com/arduino/Arduino/issues/6350 and has been solved.

Please always test using the beta or hourly build from https://www.arduino.cc/en/Main/Software before reporting issues here so that you don't report things that have already been fixed.

Regarding the warnings. That is something to discuss on the Arduino Forum:
https://forum.arduino.cc/

closed 2018-07-04 by per1234

downunder54, 2018-07-04:
OK, will do, thanks for replying, per1234!