arduino/Arduino #7757: Number of status lines.

JLNapo, 2018-07-05:
Hello,
First of all, thank you for all that you allow us to do and use.
Since the last versions of the IDE, the number of status lines is limited to a value that is too high. Why not allow as before, to let the user decide the minimum? I do not know if it's because of my machine, but I have 3 old versions with which I do not have this problem. With the version 1.8.5 I have to put "linestatus.height = -200" to be able to have a correct operation, because 1/3 of the screen it is really too much.

facchinm, 2018-07-09:
Hi @JLNapo ,
could you provide a screenshot of how the interface appears by default on your setup? Can you also provide an initial version that works file for you (like, for example, 1.6.13) so we can try bisecting the issue? Thanks!

facchinm, 2018-07-09:
It looks related with EditorConsole, so probably setting `linestatus.height` resizes also the console as a side effect. The problematic commit could be https://github.com/arduino/Arduino/commit/019022505089dd45f9beba77fc033bdc67d242e6#diff-d09f41a284a30386e40aabe70b4c9eb6 , which was applied in IDE 1.8.4.
Would you mind testing 1.8.3 to check if it is indeed the last version that doesn't show the bug? Also, maybe paste here your `console.lines` setting :wink:

JLNapo, 2018-07-09:
In fact, after some tests, I feel that version 1.8.3 does not read the "linestatus" because even if I change, it is not taken into account. It is as if the file "C: \ Program Files (x86) \ Arduino-1.8.3 \ lib \ theme \ theme.txt" was not used completely. On the other hand, the status lines have a manual correct height.

[theme(1.8.3).txt](https://github.com/arduino/Arduino/files/2175917/theme.1.8.3.txt)
[theme(1.8.5).txt](https://github.com/arduino/Arduino/files/2175923/theme.1.8.5.txt)
thanks

JLNapo, 2018-08-24:
Hello,
I thought the problem would be solved with the new version (1.8.6) but this is not the case.
Why have you touched this setting which was very good in versions below 1.8.4?
Nevertheless, I remain at your disposal to help you change that.
Thanks so much

per1234, 2018-09-26:
@JLNapo are you still having this problem? We're still waiting for you to post your `console.lines` setting. That property is in the file preferences.txt, which you can find by clicking the link at the line following **File > Preferences > More preferences can be edited directly in the file**.

per1234, 2021-03-17:
Closing due to lack of response.

closed 2021-03-17 by per1234