arduino/Arduino #7761: The menu lists are too short, will not scroll.

rootexpression, 2018-07-06:
For example Example Sketches and Boards, I have only added a few and there are now too fit in the list on screen, and they won't scroll.

This has been an existing bug for many years now, and it is a platform breaking bug, since just switching to another IDE carries many issues with it that often are not an option.

per1234, 2018-07-06:
Which OS are you using?

The menus have scrollbars for me on Windows and Linux ARM versions of the Arduino IDE. I don't have a macOS computer to test with. It's true this issue did exist in old IDE versions but that was fixed years ago.

rootexpression, 2018-07-06:
Never was fixed for me, and that spans many linux distros, including Ubuntu, Arch, Gentoo based, etc.  Its Java, so why it'd ever be any different I don't know.

per1234, 2018-07-06:
Are you using the official Arduino IDE downloaded from:
https://www.arduino.cc/en/Main/Software

rootexpression, 2018-07-06:
Yes.

facchinm, 2018-07-09:
Hi @rootexpression,
I'm on Arch too (openbox) and the scroll arrows are there for every menu (this patch for example does the trick https://github.com/arduino/Arduino/commit/bfea2724ba9e0649a1eb7c6dbd60c0e9da03d3aa ). I can try reproducing it with a bit of information, like:
- which java version is installed system-wide
- which window manager you are using
- which compositor, maybe?

Thanks!

per1234, 2018-09-26:
@rootexpression we're still waiting for you to provide the information requested by facchinm.

per1234, 2019-09-16:
Closing due to lack of response. @rootexpression if you ever feel like providing the requested information, please reply here and I'll be happy to re-open this.

closed 2019-09-16 by per1234

ijash, 2020-03-22:
> Hi @rootexpression,
> I'm on Arch too (openbox) and the scroll arrows are there for every menu (this patch for example does the trick [bfea272](https://github.com/arduino/Arduino/commit/bfea2724ba9e0649a1eb7c6dbd60c0e9da03d3aa) ). I can try reproducing it with a bit of information, like:
> 
> * which java version is installed system-wide
> * which window manager you are using
> * which compositor, maybe?
> 
> Thanks!

can this be reopened?

here's my full PC specs.

https://termbin.com/gbzn