arduino/Arduino #7763: Typo: Arduino Yún, not Yùn

hyiltiz, 2018-07-06:
I hope this page is part of the "Documentation", thus this is the correct place to report this typo. If not, feel free to tell me where. 

https://www.arduino.cc/en/Products.Compare

[image]

AFZL210, 2021-03-24:
Hi, im new to opensource may i try to resolve this typo bug

per1234, 2022-07-17:
Thanks for your report @hyiltiz!

I see that the referenced page is now no longer accessible.

The only remaining occurrence of this spelling I found was on the release blog post:

https://blog.arduino.cc/2018/03/29/arduino-yun-rev-2-is-here/

I don't know that it makes sense to edit a blog post after the fact, so I will close this as resolved. 

If anyone spots an outstanding occurrence of it in other official content, please comment here and I'll investigate.

closed 2022-07-17 by per1234