arduino/Arduino #7772: Education resources

kellySmi, 2018-07-09:
Educators do not have a clue as to where to register their kit's. They paid thousands of dollars for these kits and have instructions to register thier kit, yet a link is not found, nor is their anywhere inside their account area to allow this functionality. Where are the webinars and other info that is included? Arduino, if you want people to learn technology, teach the teachers first, by making you site easier for them.

ErnestoELC, 2018-07-09:
Hello @kellySmi 
Could you specify which Kit are you referring to? Is it the Engineering kit or CTC kit?
Thanks.

kellySmi, 2018-07-09:
The Kit is the CTC101 kit purchased from [the shop](https://store.arduino.cc/usa/arduino-ctc-101-program). The instructions for the kit a really vague, with no direct URL to register the kit or to get the resources.

charlesbones, 2018-07-10:
Hi Kelly, Thanks you so much for your feedback.

Aniram1, 2018-12-02:
anch'io ho appena ricevuto il Kit Arduino CTC 101 e all'interno della confezione ho trovato l'indicazione del sito e del codice con il quale registrarmi alla piattaforma. il sito indicato è il seguente CREATE.ARDUINO.CC/CTC/REG ma da questa pagina inserendo il codice  e username non accedo alla piattaforma  CTC 101

charlesbones, 2018-12-03:
Hi @Aniram1 
Please send us a message to ctc.101@arduino.cc with the screenshot of the error you are getting.

per1234, 2018-12-03:
@charlesbones since the work on this is being done via private communications channels, please comment here to let us know once this issue (the original one as well as @Aniram1's) can be closed as resolved.

alsalas77, 2019-05-07:
> anch'io ho appena ricevuto il Kit Arduino CTC 101 e all'interno della confezione ho trovato l'indicazione del sito e del codice con il quale registrarmi alla piattaforma. il sito indicato è il seguente CREATE.ARDUINO.CC/CTC/REG ma da questa pagina inserendo il codice e username non accedo alla piattaforma CTC 101
> 
> Google Translate:
> I have just received the Kit Arduino CTC 101 and inside the package I found the indication of the site and the code with which to register to the platform. the indicated site is the following CREATE.ARDUINO.CC/CTC/REG but from this page entering the code and username I do not access the platform CTC 101

**I am having the exact same problem.  This thread is several months old, so it seems odd to me that this is still a problem.  But considering I had so much trouble just ordering the kits in the first place, and there is no way to actually speak in real time with someone at Arduino, I guess I should not be surprised.**

charlesbones, 2019-05-08:
Hi @alsalas77

alsalas77, 2019-05-08:
Hi Charles.  While those were the instructions on the toolbox lid, the link is not correct. Here is the actual link that was sent to me from Arduino today.  Not sure why they haven't changed it either on the lid or made a re-direct to the correct page:
https://create.arduino.cc/ctc/reg/register-code

Aniram1, 2019-05-12:
Ciao, il mio problema è stato risolto.
Grazie per l'aiuto e la disponibilità.