arduino/Arduino #7773: Set timezones correctly

kellySmi, 2018-07-10:
Every date time stamp on the site/app is off by 6 hours. Just signed up for a webinar that said it was at 5:30 pm and the confirmation email says 11:30 pm EDT. Many other places had time zones off as well.

smellai, 2018-07-11:
@kellySmi can you provide a link of the website?

kellySmi, 2018-07-17:
This occurs everywhere in your site. The emails have the incorrect time zones for the webinars, or maybe no one shows up when they are supposed to. You should set the users default timezone when they signup. Every time is displayed in UTC when the time zone can be obtained from the clients browser. You have many conflicting messages about the time zones, what is central european summertime?  You might think of outsourcing your support to the US?

smellai, 2018-07-23:
Please provide ANY link. Thanks