arduino/Arduino #7775: Wrong example code for WiFi library Simple WebServer WiFi

EffC00k, 2018-07-10:
In this page:
https://www.arduino.cc/en/Tutorial/LibraryExamples
The example code for the Simple WebServer WiFi
seems to be the same as the example for the WiFi Web Server.

The former is supposed to turn on/off an Arduino LED, but it sends back
the value of 5 analog inputs just like the second example.

Also, the code does work for reporting the analog inputs, but the printWifiStatus
function is never called and the IP address does not print out, making it difficult to
know the assigned IP address.

Swapping the last two lines in setup () allows the IP address to print, making it
possible to enter the address in a remote machine's web browser.

Swapped lines:
  // you're connected now, so print out the status:
  printWifiStatus();

  server.begin();

I know this is obsolete hardware, but the examples should be correct.

per1234, 2018-07-10:
Thanks for reporting this @EffC00k

The WiFi library's SimpleWebServerWiFi example does have the expected LED code so the issue is isolated to the website:
https://github.com/arduino-libraries/WiFi/blob/master/examples/SimpleWebServerWiFi/SimpleWebServerWiFi.ino

EffC00k, 2018-07-12:
Thanks, I was able to use the github link to download the code, it works as advertised.

per1234, 2018-07-12:
We need to leave this open until the error on the arduino.cc page has been fixed.

closed 2020-10-29 by dcuartielles