arduino/Arduino #7780: Multiple Font Selection and Color issues.

pendletonrc, 2018-07-12:
Please, I have only one working eye and my other eye is not working so well. Please consider making it easier to change fonts and colors in the IDE. The default font is actually painful to read and much of the gray text is completely unreadable. I have spent most of two days fiddling with the preferences.txt and theme.txt files and I have managed to change one font face. I have learned that most of the fonts do not change even when changed either configuration file. I have learned that changing "plain" to either "bold" or "Italic" does not change the fonts to bold or italic they just seem to make the font worse.
I haven't tried to change colors yet.

I've been programming since 1972 and working on 8 bit micros off and on since the middle '70s when the 8080 came out.

per1234, 2018-07-12:
We do already have a previous request for making it easier to modify the Arduino IDE theme: https://github.com/arduino/Arduino/issues/6454

Regarding changing the font style having no effect: I can reproduce that problem with `editor.font` in preferences.txt (Windows 10 64 bit, Arduino IDE 1.8.6 Hourly Build 2018/06/18 03:33). Changing the `console.font` font style in theme.txt does work as expected (that was the only theme property I tried).

I have attempted to document the theme settings here:
https://github.com/per1234/ThemeTest/blob/master/README.md
maybe that will be helpful to you

pendletonrc, 2018-07-13:
Thank You for looking at this. I am using 1.8.5, on Ubuntu 18.04 64. It
does not work for me.

per1234, 2019-02-02:
We now have a dedicated issue for the font style parameter of `editor.font` having no effect (https://github.com/arduino/Arduino/issues/8484), and already had one about making the theme easier to edit, so I'm going to close this issue. We'll track progress in resolving the problems you reported in the other issue reports.

closed 2019-02-02 by per1234