arduino/Arduino #7786: Arduino IDE 1.8.5 freeze during start

Rasputinbkz, 2018-07-15:
Hi

I am facing issues trying to run the latest IDE. I am on a linux mint 64 and the IDE is 1.8.5. When it starts, I get, installing packages; preparing cards and Starting. It stays forever in this state, I need to kill the java application to close it...

When I run it on the terminal I only get:
Picked up JAVA_TOOL_OPTIONS: 
and then it stays like this forever...

Any clue?

Thanks

facchinm, 2018-07-16:
Hi @Rasputinbkz ,
can you check if you have any symlink loop inside your sketchbook? The IDE may be scanning the folders forever to list installed libraries. To make sure this isn't the case, create a folder called `portable` on the top level of the archive you extracted (eg `arduino-1.8.5/portable`) and try launching the ide. Let me know if it works!

Rasputinbkz, 2018-07-17:
Worked like a charm. Thanks Facchinm

closed 2018-07-17 by facchinm