arduino/Arduino #7789: Arduino Create - Cannot compile for CPX?

ladyada, 2018-07-16:
seeing this issue come up lately, i can reproduce that code does not compile in latest Create

https://twitter.com/STEM_ROX/status/1018583757475975168

[image]

facchinm, 2018-07-17:
Hi @ladyada ,
the problem is related with the library itself (I don't know why we didn't spot it before).
On AVR, no problem at all. On ARM (CPX) [this](https://github.com/adafruit/Adafruit_CircuitPlayground/blob/master/utility/ffft.S) file is empty and the assembler has no clue on how to compile it, so it fails.

Simple patch: add this snippet at the end of ffft.S
```
#ifdef __SAMD21G18A__
.cpu cortex-m0plus
.fpu softvfp
#endif
```

I'd close the issue here and move it to https://github.com/adafruit/Adafruit_CircuitPlayground since it's not really a Create bug.

closed 2018-07-17 by facchinm

ladyada, 2018-07-17:
ok fixed and bumped!
https://github.com/adafruit/Adafruit_CircuitPlayground/releases/tag/1.8.1