arduino/Arduino #7795: Adding SPI programming on a Raspberry Pi (from the IDE)

lewispg228, 2018-07-18:
I've been working on a [stand-alone AVR programming solution](https://learn.sparkfun.com/tutorials/raspberry-pi-stand-alone-programmer) for SparkFun over the past couple years. Currently, we use Python (watching a GPIO) to engage a bash file that calls avrdude.  We have found that the SPI hardware built into the raspberry pi is very reliable and it would be awesome to be able to also develop within the arduino IDE (on a raspi), and upload via SPI.

Would this be possible? Is it a crazy request?

per1234, 2018-07-19:
I have a semi-working proof of concept. Maybe you could take a look at it and we can discuss it further in this thread:
https://github.com/per1234/Arduino_Boards/pull/1

It's currently implemented in SparkFun AVR Boards but the change could be applied to Arduino AVR Boards instead if this is something Arduino wants.

facchinm, 2018-07-19:
Hi @lewispg228 ,
if you switch to latest avrdude ([6.3.0-arduino14](http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino14-armhf-pc-linux-gnu.tar.bz2)) which is included in the hourly build of the IDE you'll get `linuxspi` programmer support for free :wink: 
You only need to specify where the reset pin is connected on the RPi gpios in your `avrdude.conf` like this:

```
programmer
  id = "linuxspi";
  desc = "Use Linux SPI device in /dev/spidev*";
  type = "linuxspi";
  reset = 67;
;
```

Once added, you can program the Arduino with `avrdude  -c linuxspi -P /dev/spidev0.0 ... `

per1234, 2018-09-26:
Since there is some activity on this topic, I wanted to leave a note in case it might help someone else avoid the confusion I initially suffered:

With the current 6.3.0-arduino14 version of AVRDUDE provided by Arduino, use of the linuxspi programmer will fail if run as non-root. This issue has since been fixed by https://github.com/facchinm/avrdude/pull/2 but Arduino has not done an AVRDUDE release since that time.