arduino/Arduino #7799: [Library Manager] Please add this lovely library 💖

bblanchon, 2018-07-19:
Hello Arduino team,

I see that you have a long backlog of libraries to add, but I hope that you'll find the time to add this new library of mine:

* Repository: https://github.com/bblanchon/ArduinoTrace.git
* Homepage: https://github.com/bblanchon/ArduinoTrace

It should be ready to go:

- [X] git tag
- [X] `library.properties`
- [X] examples
- [X] `keywords.txt`

Thank you very much 😊😊😊

cmaglie, 2018-07-26:
Done, sorry for the delay

closed 2018-07-26 by cmaglie

kvonarya, 2020-02-16:
This is to indicate TRACE, DUMP and <ArduinoTrace.h> have no value and are of no help troubleshooting Arduino IDE sketches when the first thing that fail is serial monitor displaying texts.
Serial monitor freezes and stops while my program keeps running and functioning as can be seen thru
Arduino UNO I/O.
I do nothing fancy or special with serial monitor, all related to serial in my code is like:
.....
#include "Arduino.h"
#include "SoftwareSerial.h"
.....
SoftwareSerial mySoftwareSerial(10, 11); // RX, TX
.......
void setup() {
  mySoftwareSerial.begin(9600);
  Serial.begin(9600);
  Serial.println();
  ......
}
//********************** setup ends here ******************************************

//************************************ Main Loop **********************************
void loop() {
  .........
  Serial.println("write message 1");
  .........
  Serial.print("write message 2 ");
  ..........
  Serial.println(avariable);
  ..........
  Serial.println("write message 3");
} 
So no matter where I put TRACE or DUMP, I see nothing on serial monitor.