arduino/Arduino #7825: [Library Manager] Please add BME280 Sensor Library

flybrianfly, 2018-07-26:
Please add the BME280 sensor library to the library manager:
https://github.com/bolderflight/BME280

per1234, 2018-07-26:
There is already a library named `BME280` in the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json). Each library must have a unique name. Please do the following:
1. Change the `name` value in your library.properties file to a unique name. You can check which names are taken in the Arduino IDE at **Sketch > Include Library > Manage libraries** or by searching the contents of the [json file](http://downloads.arduino.cc/libraries/library_index.json).
1. Redo the existing v1.0.0 tag or update the `version` value in library.properties and create a new tag/release.
1. Comment here so we'll know we can proceed.

flybrianfly, 2018-07-26:
Thanks, the name and version has been updated.

cmaglie, 2018-08-08:
Done, sorry for the delay!

closed 2018-08-08 by cmaglie