arduino/Arduino #7836: [Library Manager] Please add ArduinoESPAT library

TakumiShinoda, 2018-07-28:
Hello!

ArduinoESPAT library can link Arduino with esp8266 by AT command.
Could you add our ArduinoESPAT library to library manager?
url: https://github.com/nyampass/ArduinoESPAT-Library

 Thanks.

per1234, 2018-07-28:
Your library.properties is missing required fields:
- `paragraph`
- `url`

Please resolve this issue and then either create a new release/tag (after updating the `version` field in library.properties) or else redo the existing tag.

After that is done please comment here so we know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

TakumiShinoda, 2018-07-28:
Edit done!
I added url and paragraph at library.properties.

I'm sorry for taking more time, If I still have problem.
Thank you for taking time.

cmaglie, 2018-08-08:
Done, sorry for the delay!

closed 2018-08-08 by cmaglie