arduino/Arduino #7839: Menu fonts broken with unrecognizable characters

kongwy, 2018-07-29:
Run on Windows 10 (build 17134) with Chinese language, menu fonts seem broken with unrecognizable characters, I tried to change IDE language setting to English and UI font configuration in `lib\theme\theme.txt` but still not work.

IDE menu display with default (Chinese) language setting:
[image]

IDE menu display with English language setting:
[image]

Settings windows and other text can be displayed in normal:
[image]

facchinm, 2018-08-20:
HI @kongwy ,
the font in use is fairly different from the "standard" Windows one which looks more like
[image]
Your system probably needs a font (not present by default with Chinese installation? ). Does garbled text also appear on a fresh [portable](https://www.arduino.cc/en/Guide/PortableIDE) installation?

kongwy, 2018-08-20:
Hi, @facchinm thanks for your help.

This problem appears in both instances installed by .exe or unzip from the compressed file. I tried changing font config in `lib\theme\theme.txt` but didn't help.

As you said, I just used Beyond Compare to compare the `Fonts` folder from my current Windows install and the one extracted from the `install.wim` from a Windows (Simplified Chinese Pro Edition) ISO. Basically, there are no differences except four TTF files of Yu Gothic have a newer version on my computer.

facchinm, 2019-09-20:
@SimonePDA  we need to find a place to add the documentation about the fix (which is here https://github.com/facchinm/Arduino/commit/fb89a670783d8a45db4998d43ea302ac63147793 ) but can't be applied for everyone or the render would be terrible. Any hint?

SimonePDA, 2019-09-20:
@facchinm I already opened a task on Jira because with the new release of the IDE, the existing documentation is definitely obsolete. Maybe this information could be added to the refreshed "environment" docs. However I have no time schedule for that as it is in FT-Content hands for assignment and priority.
We document the IDE in these two places:
https://www.arduino.cc/en/guide/environment
https://create.arduino.cc/projecthub/Arduino_Genuino/getting-started-with-the-arduino-software-ide-623be4

closed 2020-09-29 by dcuartielles

per1234, 2020-10-01:
>we need to find a place to add the documentation about the fix

The workaround has now been documented:
https://www.arduino.cc/en/Guide/Troubleshooting#why-is-the-arduino-ides-menu-text-gibberish