arduino/Arduino #7842: [Library Manager] Please add ArduinoThingsOfValueSDK library

kyuseok-oh, 2018-07-30:
Hello,

Please Add ArduinoThingsOfValueSDK library to library manager.

Here is URL : https://github.com/kyuseok-oh/ArduinoThingsOfValueSDK

Thank you!!!

per1234, 2018-07-30:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

kyuseok-oh, 2018-07-30:
Hello,
I just finished creating a release.
Please check it.
Thank you.

cmaglie, 2018-08-08:
Done!

closed 2018-08-08 by cmaglie