arduino/Arduino #7843: Arduino Windows IDE 1.8.6 Hourly

vangalvin, 2018-07-30:
Not sure where to post this but an interesting issue with the build I am using.

If open an .ino file by downloading in chrome, then click on the downloaded file it will say it needs to be saved in a folder. You do that and it saves is in the Downloads dir. then proceeds to copy all the content of the Downloads DIR in to the folder in created. The process that does this seems to be hidden and the only sign of it is the computer starts to run like a slug.

Only option to terminate this seems to be to reboot.

per1234, 2018-07-30:
That new behavior is caused by https://github.com/arduino/Arduino/pull/6404

cmaglie, 2018-08-08:
I'm actually wondering if the IDE should do this non-trivial operation automatically.
Maybe it's better if the IDE just refuse to open the sketch and display a message saying:

`The sketch Blink.ino must be in a folder named 'Blink'`

per1234, 2020-10-05:
#6404 was reverted so I'm closing this as fixed.

closed 2020-10-05 by per1234