arduino/Arduino #7846: Arduino 1.8.5 does not work after install

ixb11170, 2018-07-30:
The application does not start after install on RPI 3b running Raspbian Jessie.

I followed the instructions on https://www.arduino.cc/en/Guide/Linux, downloaded the ARM version, unpacked, built and installed all successfully.

I got the warning:

    Adding desktop shortcut, menu item and file associations for Arduino IDE...
    touch: cannot touch '/root/.local/share/applications/mimeapps.list': No such file or directory
    /usr/bin/xdg-mime: 803: /usr/bin/xdg-mime: cannot create /root/.local/share/applications/mimeapps.list.new: Directory nonexistent
    done!

Found this message in a few other issues posts but can't resolve the problem.

I tried to start the application from the Desktop, as well as from the command line and neither worked. Let me know if any more details would help.

Cheers.

cmaglie, 2018-08-08:
Please:
- un-tar the arduino archive in a folder
- run `arduino` in that folder from command line
- copy the output here

dsyleixa, 2018-08-09:
for me the installation worked this way:
wget http://downloads.arduino.cc/arduino-1.8.5-linuxarm.tar.xz
tar -x -f arduino-1.8.5-linuxarm.tar.xz

no automatic arduino link creation, instead localize the arduino program in the arduino directory by leafpad and then create a link on the desktop

AndersonLovesYou, 2020-06-23:
Hello, I had a same issues with a Raspberry Pi 4, Quad core 64 bit ARM cortex-A72 cpu.

I followed this tutorial exactly.
https://magpi.raspberrypi.org/articles/program-arduino-uno-raspberry-pi

using sudo apt-get install arduino will install a very outdated version of Arduino on a computer.

Executing the command will result in following error messages.

root@raspberrypi:/opt/arduino-1.8.13# sudo ./install.sh
Adding desktop shortcut, menu item and file associations for Arduino IDE...

touch: cannot touch '/root/.config/mimeapps.list': No such file or directory
/usr/bin/xdg-mime: 848: /usr/bin/xdg-mime: cannot create /root/.config/mimeapps.list.new: Directory nonexistent

 done!

Arduino won't run if you click on the GUI.

Running ARDUINO will result in an error from a source code.

root@raspberrypi:/opt/arduino-1.8.13# **./arduino**
./arduino: line 35: /opt/arduino-1.8.13/java/bin/java: cannot execute binary file: Exec format error

I am down to help if there are any questions but I am a NOOB.

facchinm, 2020-06-23:
Hi @AndersonLovesYou , depending on your Raspbian distribution you should use the arm32 or the arm64 build. The processor is indeed 64bit but the RaspberryPi foundation used to released a single version of their software that runs on all boards (so it's 32bit).
Either way, if you downloaded the 64 bit IDE try the 32 bit one :slightly_smiling_face:

Eewec, 2020-11-14:
As a heads up, I just downloaded the 32 bit version onto my 32 bit RaspberryPi. The install script still has this issue and refuses to play nice.

arturo-rodriguez35, 2021-04-12:
> Please:
> 
>     * un-tar the arduino archive in a folder
> 
>     * run `arduino` in that folder from command line
> 
>     * copy the output here

[work@haco01 arduino-1.8.13]$ ./arduino
./arduino: line 35: /home/work/Documents/Software/arduino-1.8.13/java/bin/java: cannot execute binary file
Any ideas?

dsyleixa, 2021-04-12:
look above:
https://github.com/arduino/Arduino/issues/7846#issuecomment-411849921

arturo-rodriguez35, 2021-04-12:
Still doesn't work for me. I started from zero (deleted all previous installation, downloaded, untar, and ran) and still get the same message

dsyleixa, 2021-04-12:
what exactly was your download command?

magellannh, 2021-07-03:
I ran into the same problem using a Raspberry Pi 4 Model B Rev 1.4 running Debian 10.9 (buster).

I was able to correct the error with the following command sequence
sudo su
cd /root
mkdir .config
chmod 777 .config (this is probably not needed)
exit
sudo /opt/arduino-1.8.15/install.sh

I didn't uninstall after the initial error.  I just entered these commands exactly as shown above and the installer executed without any errors.  

After that, I was able to successfully launch the IDE using the Programming->Arduino IDE menu option.

magellannh, 2021-07-03:
Also note that I encountered this problem using 1.8.15, (latest as of today) and not 1.8.5