arduino/Arduino #7861: exception thrown when font change with [ctrl+wheel] and a mouseMoved event happens

Vipas-ana, 2018-08-04:
### Version
Arduino 1.9.0 beta 78
### OS
Mac OS 10.11.6
### Behavior
Use the [ ctrl+scroll wheel ] to change the font size in the SketchTextArea, making sure not to move the mouse while doing so.  If you then do the same while moving the mouse, this exception is thrown.
 ```
Exception in thread "AWT-EventQueue-0" java.lang.ArithmeticException: / by zero
	at org.fife.ui.rsyntaxtextarea.SyntaxView.viewToModel(SyntaxView.java:949)
	at javax.swing.plaf.basic.BasicTextUI$RootView.viewToModel(BasicTextUI.java:1552)
	at javax.swing.plaf.basic.BasicTextUI.viewToModel(BasicTextUI.java:1101)
	at javax.swing.plaf.basic.BasicTextUI.viewToModel(BasicTextUI.java:1074)
	at javax.swing.text.JTextComponent.viewToModel(JTextComponent.java:1395)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.viewToToken(RSyntaxTextArea.java:3166)
	at processing.app.syntax.SketchTextArea$SketchTextAreaMouseListener.mouseMoved(SketchTextArea.java:352)
	at java.awt.AWTEventMulticaster.mouseMoved(AWTEventMulticaster.java:330)
	at java.awt.AWTEventMulticaster.mouseMoved(AWTEventMulticaster.java:329)
	at java.awt.Component.processMouseMotionEvent(Component.java:6578)
	at javax.swing.JComponent.processMouseMotionEvent(JComponent.java:3342)
	at java.awt.Component.processEvent(Component.java:6302)
	at java.awt.Container.processEvent(Container.java:2238)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2296)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4897)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4475)
	at java.awt.Container.dispatchEventImpl(Container.java:2282)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

facchinm, 2018-08-20:
I tried to dig deeper into this bug once, but it's not easily solvable by our side (rsyntaxtextarea does a lot of math but no obvious division by zero). I've only experienced it when zooming in and out compulsively, so it may be related with timings and events.

Vipas-ana, 2018-08-28:
This may be interesting...
I noticed it will also happen by simply using the key command for changing font size while you  move the mouse. Interesting thing is that when the shift modifier is used, in the case of "Increase Font Size", this exception is not thrown.  Also, if you add the shift modifier to the use of "Decrease Font Size" the exception is not thrown.

(no more just from compulsive scrolling) :wink:

cmaglie, 2018-08-28:
may be related: https://github.com/bobbylight/RSyntaxTextArea/issues/257

Vipas-ana, 2018-08-28:
Hope this helps..

Vipas-ana, 2018-09-09:
Can we get this merged? :)

facchinm, 2018-09-10:
It would be cool if @bobbylight could release a bugfix-only version of RSyntaxTextArea instead than forcing us to self compile it from git head :wink:

bobbylight, 2018-11-25:
Until I create a new release (should be before the end of the year), you can use the 3.0.0-SNAPSHOT version from the [Sonartype maven repo](https://oss.sonatype.org/content/repositories/snapshots/com/fifesoft/rsyntaxtextarea/).

Vipas-ana, 2019-01-17:
Looks like we can now bundle the binary.  @facchinm can we get this bundled in an hourly build?

facchinm, 2019-01-18:
@Vipas-ana sure, thanks for the heads up!

commit f15ddeb8 (2019-01-18): Update rsyntaxtextarea to 3.0.2

The name is misleading since 3.0.2 had been officially released (but the filename is totally wrong).
The jar comes from https://netcologne.dl.sourceforge.net/project/rsyntaxtextarea/rsyntaxtextarea/3.0.2/rsyntaxtextarea-3.0.2.zip

Fixes #7861
closed 2019-01-18 by facchinm