arduino/Arduino #7863: Sketchbook setting lost when Linux mount fails

bphermansson, 2018-08-05:
I have my Sketchbook folder on a NAS and my Linux computers don't always mount the folder as they shall when it starts. If I then launch Arduino IDE it tells me that the Sketchbook is lost and changes the setting for the Sketchbook folder. I think this behavior is wrong, why not just tell be the Sketchbook folder is missing and let the setting be? I can then correct the mount issue without going through the trouble of changing the settings.

facchinm, 2018-08-20:
Hi @bphermansson ,
the IDE tends to rewrite `preferences.txt` way too often than needed, and this provokes other unintended behaviours (since every time a Preferences structure field is changed it tries to update the backing storage). Handling this in a saner way could be a nice (and not so demanding, if anyone in the community wants to tackle it  :wink: ) task

facchinm, 2018-08-20:
https://github.com/arduino/Arduino/pull/7879 could help too in your case (not a perfect fix  for your specific problem, indeed, since it would also remove some functionalities like "recent sketches")

bphermansson, 2018-09-26:
Hmm I certainly won't let recent sketches go, that's way to useful. I maybe take a look at the code and see if I have any ideas of improvements.